ksheumaker/homeassistant-apsystems_ecur #162: v1.2.27 did not work

12christiaan, 2023-03-01:
After a long time i had some time to look at this intgration again.
did update from v1.2.21 > v1.2.27 (latest). 
but that did not work, keep getting this error.
```
Mar  1 11:56:47 homeass hass[686]: 2023-03-01 11:56:47 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry ECU: 215000012502 for apsystems_ecur
Mar  1 11:56:47 homeass hass[686]: Traceback (most recent call last):
Mar  1 11:56:47 homeass hass[686]:   File "/srv/homeassistant/lib/python3.10/site-packages/homeassistant/config_entries.py", line 353, in async_setup
Mar  1 11:56:47 homeass hass[686]:     result = await component.async_setup_entry(hass, self)
Mar  1 11:56:47 homeass hass[686]:   File "/home/homeassistant/.homeassistant/custom_components/apsystems_ecur/__init__.py", line 210, in async_setup_entry
Mar  1 11:56:47 homeass hass[686]:     await hass.config_entries.async_forward_entry_setups(config, PLATFORMS)
Mar  1 11:56:47 homeass hass[686]: AttributeError: 'ConfigEntries' object has no attribute 'async_forward_entry_setups'
```
when i loaded v1.2.26, i get data into the system again. No Error. seams better.

Doudou14, 2023-03-05:
Do you have the latest version of Home Assistant?

HAEdwin, 2023-04-17:
Hopefully a deinstall and install did the job - I'm closing this issue since there was no response for over a month. Might any issue arrise, feel free to open a new issue.

closed 2023-04-17 by HAEdwin