ksheumaker/homeassistant-apsystems_ecur #163: Needs daily HA restart

dev-github-ha, 2023-03-17:
Love this integration however it seems to hang up and stop recovering data and requires me to restart home assistant to get it working again.  Is there any fixes that don’t require a HA reboot?

musing an ecu-r

jensihnow, 2023-03-17:
Experienced the same on an ECU-B, but it got better recently. 

Firmware: ECU_B_1.2.26

HAEdwin, 2023-03-18:
Recently, different results due to ECU firmware upgrades have been noticeable. Unfortunately, APsystems does not provide us with insight into release notes so that we can anticipate them. for me personally (ECU-R 2160------) the integration still works perfectly. For ECU-Rs starting with 2162----- the automatic reboot should still work, but in general it remains a questionable matter until APsystems provides a clear insight. I would advise trying a modbus integration if your ECU should be compatible with it, that is the official method to get data from the ECU yourself. Experiments with a MITM proxy currently give the impression that this cannot be applied generically for each ECU model either. It is known that blocking APsystems domains results in a non working state for the ECU.

Doudou14, 2023-03-18:
Since my ECU-B was updated to Version 1.2.26, I lose Zigbee signal with the inverter (For 5min to more than 60min at least once a day).
On the APS site I have no hole in the production. But I think the Zigbee problem is with the latest software but the inverter is sending the cached data to the ECU when communication is restored.

HAEdwin, 2023-03-19:
You say "I lose Zigbee signal with the inverter", this would mean the ECU would detect it as being offline. If this is true, try to place the ECU elsewhere in your home and look at the signal strength of the inverters in the integration. Your ECU-B must also have a stable WiFi connection. Indeed "missed" data is retransferred to the EMA site in de evening because the ECU holds the data for a period of time. It would be a pity if new firmware leads to less connectivity but other new zigbee devices of WiFi devices nearby could also be the cause.

Doudou14, 2023-03-20:
That's right, the online inverter goes to 0. It's been several months that everything worked and since the update of the ECU B, I have this problem.
I have 70% on the zigbee signal.
Maybe the new firmware updated the inverter.

Doudou14, 2023-03-28:
I think the Zigbee problem comes from a disturbance because my philips hue also have problems at the same times

HAEdwin, 2023-03-31:
@jensihnow and @dev-github-ha If the ECU is a set number of times unreachable, the "ECU query device" switch might have flipped to the off position. Then after having solved the issue with the ECU (power cycle) don't forget to flip the switch to the On position again. Same is achieved with a HA reboot but that really is not needed. For later ECU-R models starting with 2162------ and ECU-C models the reboot is automatically after the set number or retries.

HAEdwin, 2023-10-25:
I'm closing this topic because no more responses have been received. If another issue arises, please feel free to open a new issue.

closed 2023-10-25 by HAEdwin