ksheumaker/homeassistant-apsystems_ecur #164: Long term statistics for inverter sensors

mcflexus, 2023-03-24:
I see the ecu sensors have long term statistics enabled but the inverter sensors don't. Can you enable them? I think especially for temperature sensor it makes much sense to do so. Thank you for this great integration!

HAEdwin, 2023-03-24:
They are long term (at least when I quick tested it)
[image]

mcflexus, 2023-03-24:
I mean something else, this is just the history of a few days, long term statistics are stored for ever. See https://data.home-assistant.io/docs/statistics/

HAEdwin, 2023-03-24:
I see what you mean, state_class is missing. Are you able to add that to the integration?

mcflexus, 2023-03-24:
Don't think that should be a problem. I'll try to find some time this weekend.

mcflexus, 2023-03-25:
just made a pull request, tested locally and working as expected.

HAEdwin, 2023-03-25:
Merged the pull request, thank you!

closed 2023-03-25 by HAEdwin

mcflexus, 2023-03-25:
No problem!