ksheumaker/homeassistant-apsystems_ecur #166: Can't add sensor.ecu_current_power to energy dashboard HA

m3tc0n, 2023-03-29:
Hello,

I have successfully installed the APsystems integration via HACS. 

Now I would like to add the sensor 'sensor.ecu_current_power' to the energy dashboard, but it is not displayed as an option (only the sensors sensor.ecu_lifetime_energy and sensor.ecu_lifetime_energy are displayed). 

How can this be resolved? 

Thanks in advance.

rivendellstuff, 2023-03-29:
same issue here.
I don't want to continue to add the DAILY kw to the animated graph, I want to know the CURRENT power at all times coming from solar. 

I'd like to create this Tesla graph but it seems impossible at the moment: 

https://user-images.githubusercontent.com/128875345/228467330-c5f2d47b-bb27-4e61-9201-643eef9d774c.mp4

also the inverter **temperature** sensors appear to not be usable? Even though the details of them do show a number, a thermostat icon and even C as the measurement. When I add it to any dashboard the value is just unknown and the actual temp is not a selectable attribute.

HAEdwin, 2023-03-29:
https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/107#issuecomment-1247191667

HAEdwin, 2023-03-30:
You can use the integrations entities to create seperate bar graphs. The way the Energy Dashboard functions cannot be influenced by this integration. Try the homeassistant.io forum and makers of the Energy Dashboard. If any other question arrise about this integration, feel free to open a new ticket.

closed 2023-03-30 by HAEdwin

HAEdwin, 2023-03-30:
No reason to get upset. Current power is in (W) but the Energy Dashboard displays Energy Distribution in (Kwh) for all entities only, that is by design, I can't help that. The animated flows of energy after sundown is also by design. Hope things are clearer now.

rivendellstuff, 2023-03-30:
> No reason to get upset. Current power is in (W) but the Energy Dashboard displays Energy Distribution in (kWh) for all entities only, that is by design, I can't help that. The animated flows of energy after sundown is also by design. Hope things are clearer now.
> 
> Maybe this will help you: https://community.home-assistant.io/t/wth-doesn-t-the-energy-dashboard-work-with-watts-as-a-source/474891

ill read the thread over tomorrow and see if its helpful.  i guess we have to accept that real-time monitoring of hydro is something HA cant do yet. It just seems sad that someone like you and others put a lot of work into these integrations and the sensors are RIGHT there....just not usable.  I wasnt upset but I can tell you as this type of setup becomes more an more popular you're gonna get this question here more and more. Where I live they're talking about cutting natural gas completely in the next ten years. Solar and battery are gonna blow up. When a setup like this costs 60K+... ppl are gonna wanna monitor it and see the metrics. Like I said.. the numbers are already there... it like hurts that we cant just use them how we want to.

Guess we're stuck with the crappy tesla app.. so much for automation and having all the things in one place.

thanks for trying anyway.

m3tc0n, 2023-03-30:
@HAEdwin maybe a stupid question, but is it not possible to change the current power consumption from W to kWh in your integration? I guess that should solve the problem with integrating the sensor in the energy dashboard.

HAEdwin, 2023-03-30:
@m3tc0n Yes, I could if it doesn't create a dependency with another integration like Riemann but you'll have to provide a formula which I can then add to the new sensor. Easier option is to create the new sensor in configuration.yaml like explained in this thread https://community.home-assistant.io/t/wth-doesn-t-the-energy-dashboard-work-with-watts-as-a-source/474891

rivendellstuff, 2023-04-03:
I ended up building a custom one from scratch (graphic animations and all) 
If anyone is interested I'll share the zip with some basic instructions.  lmk

https://user-images.githubusercontent.com/128875345/229443038-aee4f1dc-f205-4850-b25c-e2ac86ba37ef.mp4

m3tc0n, 2023-04-03:
Nice, I'm interested. Thanks!

rivendellstuff, 2023-04-03:
> Nice, I'm interested. Thanks!

alright cool. i gotta work in a custom template cuz theres 9 variations of energy flow. i need to work in swapping 1 of 9 images out based on source energy. it sorta works now but not perfect. i might need to integrate a custom template thing which make the instructions more complicated....

once i have that working right illl post it. unless you wanna do that work?  then i can give u the images. .. lol... but im bust with work this week so probably next. stay tuned

rivendellstuff, 2023-04-14:
> Nice, I'm interested. Thanks!

I put it up [here](https://github.com/rivendellstuff/homeassistant-lings-animated-power-graph). I'm still tweaking it but if you wanna try it now as it is have a go. 

Took a bit of time to get all the math right. I'm still tweaking it as its my first pass at anything home assistant. I'll probably refine it over time but it works for now. See how you get on. It looks more daunting than it is... but it could be plug and play mostly if you have the same hardware as me. If not you'll need to rename a few core sensors (and ids on the inverters). See the readme.

There's also an APSystems inverter output and temperature dashboard I made included as well.
Let me know how it goes if you want > in the repo i posted above so that the author of this one doesnt keep getting notifications here with our chat :)

**What it looks like:**

https://user-images.githubusercontent.com/128875345/231962974-83bc04b1-6093-4a9e-aa31-6ee98177cab8.mp4

[image]

[image]