ksheumaker/homeassistant-apsystems_ecur #169: get a daily "connection refused" only a power cycle resolves the issue

TimWinkler, 2023-04-02:
Logger: homeassistant.helpers.frame
Source: helpers/frame.py:77
First occurred: 1:15:20 PM (1 occurrences)
Last logged: 1:15:20 PM

Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for apsystems_ecur using this method at custom_components/apsystems_ecur/__init__.py, line 199: hass.config_entries.async_setup_platforms(config, PLATFORMS)

TimWinkler, 2023-04-02:
Home Assistant 2023.3.6 Supervisor 2023.03.3 Operating System 9.5 Frontend 20230309.1 - latest

HAEdwin, 2023-04-02:
https://github.com/ksheumaker/homeassistant-apsystems_ecur/releases/tag/v1.2.27 should solve this issue.

closed 2023-04-02 by TimWinkler