ksheumaker/homeassistant-apsystems_ecur #173: ECU reset time

Robindeko, 2023-04-06:
First of all: Thank you for this integration, ik works great.

But since the last update, I seem to have a slight problem.
The "Today Energy" sensor is not reset at 00:00 my time, but at 02:00
My guess is that it resets at UTC time.

This results in my energy dashboard starting with a negative energy value every day.
Can this be fixed?

Thanks in advance!
[image]

TexTown, 2023-04-06:
In the Energy Dashboard, you should be using the 'ECU Lifetime Energy', not the 'ECU Today Energy'.

Another observation: My 'ECU Today Energy' used to reset at app. 02.00 AM as well, but since the latest firmware (now at ECU_R_1.2.24) this varies. Sometimes it resets at 22:00, sometimes later, today at 02:15, very inconsistent.

Robindeko, 2023-04-06:
Ah, I did not realize the ECU resets this counter.
I will use the lifetime Energy.
Thanks for the quick response!

closed 2023-04-06 by Robindeko