ksheumaker/homeassistant-apsystems_ecur #174: negative energy

wallb35, 2023-04-06:
Hello,
last two days 5 and 6 April 2023, I've got past day energy produced as a negative value in Energy Dashboard using ECU today Energy as Solar Production sensor (see two screenshots). Was working for month before.
Between 2 and 3 am in Dashboard energy, probably when the sensor reset in my ECU-B, all previous production from passed day come as negative value.
no problem with ECU total energy but with less precision.
looks like the sensor have the good value (see attached file).
I reboot my ECU-B today. Need to wait for tomorrow to see if it solved my issue.
ECU-B 1.2.26
ECU Integration 1.2.28
Home assistant container 2023.3.0
Regards.
Yannig
[image]
[image]
[image]

Doudou14, 2023-04-06:
I have the same problem, so I use "sensor.ecu_lifetime_energy". But I don't see less precision.

HayaminiNL, 2023-04-06:
I am experiencing the same thing with the integration:
[image]

It gives the negative number of the amount of kW of the previous day.

wallb35, 2023-04-06:
> I have the same problem, so I use "sensor.ecu_lifetime_energy". But I don't see less precision.

lifetime energy is one digit precision vs today 2 digit precision.

HayaminiNL, 2023-04-06:
Just my 2 cents on the code change 3 days ago.

The 'sensor' that stores the value is using another value as its input:
[image]

It is changed to: "stateclass=SensorStateClass.TOTAL" from "stateclass=SensorStateClass.TOTAL_INCREASING".

Before the change both sensors use the TOTAL_INCREASING.
Maybe this got mixed up, so that today_energy should use TOTAL_INCREASING and lifetime only TOTAL?

Does it make sense when I thing that it caused the 'total' going negative when being reset to zero at the new start of the day-measurement?

renini, 2023-04-06:
Experienced exactly the same.

ViperRNMC, 2023-04-07:
Same issue here, started from 5april

HayaminiNL, 2023-04-07:
Swapping the variables fixed the issue on my side.

When you connect via Terminal to the HASS node, you can edit the PY file to make it work for now.

wallb35, 2023-04-07:
> Swapping the variables fixed the issue on my side.
> 
> When you connect via Terminal to the HASS node, you can edit the PY file to make it work for now.

Thank you, modification done. Waiting for result tomorrow morning ;)

HayaminiNL, 2023-04-08:
Well done for this quick merge!

@wallb35, did the code change result in 'normal operation for you too?

wallb35, 2023-04-08:
I forgot to restart home assistant :/ so no change yet. Hope this will be good tomorrow.

wallb35, 2023-04-09:
Latest version solved the problem. Fixed

Eraser127, 2023-04-09:
Same problem here with ECU-R, so just need to edit that sensor.py? Or will there be a automatic update?

HAEdwin, 2023-05-07:
I guess this is fixed so I'll close the issue. Feel free to open a new issue if needed.

closed 2023-05-07 by HAEdwin