ksheumaker/homeassistant-apsystems_ecur #178: howto install apsystems_ecur on home assistant core

soundart, 2023-04-15:
Hi,

I would like to install this component on home-assistant core and got stuck.

The problem: I have an installation on my router where home-assistant is installed via pip. (I think this is the "core" version)

Somehow my setup looks different compared to the instructions given in the readme. Basically I do not have the HACS thing and I am reluctant trying to add that too, as the router is quite resource constrained already.

I have here:

homeassistant@home-assistant:/srv/homeassistant$ ./bin/pip list | grep assis
home-assistant-frontend 20220504.1
homeassistant           2022.5.5

and an ECU-B at 192.168.168.130:

`$ nc -v 192.168.168.130 8899
Connection to 192.168.168.130 8899 port [tcp/*] succeeded!
APS1100160001END
APS11009400012163000678320110012ECU_B_1.2.26009Etc/GMT-8        VEND
`

What I tried:

I cloned the https://github.com/ksheumaker/homeassistant-apsystems_ecur repo and 
copied the  folder homeassistant-apsystems_ecur/custom_components/apsystems_ecur to the custom_components folder below the config_dir of hass (that is  here /home/homeassistant/.homeassistant/custom_components/apsystems_ecur).

Then I started changed configurations.yaml:

`apsystems_ecur:
    host: 192.168.168.130
    scan_interval: 60
` 
and started hass from the shell:

` /srv/homeassistant/bin/python3 /srv/homeassistant/bin/hass -c /home/homeassistant/.homeassistant -v --debug --log-file=./hass.log`

I see in the log some apsystems_ecur related messages, but somehow I expected sensors to appear:

`2023-04-15 07:44:32 INFO (SyncWorker_0) [homeassistant.loader] Loaded apsystems_ecur from custom_components.apsystems_ecur
2023-04-15 07:44:32 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration apsystems_ecur which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
...
2023-04-15 07:44:37 INFO (MainThread) [homeassistant.setup] Setting up apsystems_ecur
2023-04-15 07:44:37 INFO (MainThread) [homeassistant.setup] Setup of domain apsystems_ecur took 0.0 seconds
2023-04-15 07:44:37 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event component_loaded[L]: component=apsystems_ecur>
...
2023-04-15 07:44:44 DEBUG (Recorder) [homeassistant.components.recorder] Processing task: EventTask(event=<Event component_loaded[L]: component=apsystems_ecur>)
` 

Is it possible to run this extension with the homeassistant core version?

HAEdwin, 2023-04-15:
Looks like you're missing the integration config entries which are stored in /config/.storage/core.config_entries. Normally after (manual/HACS) installation you add the integration to enable the integration to initialize and store discovered entities. I'm not sure how you can add them in your case. Here is an example of how it looks like in the core.config_entries file:
[image]

soundart, 2023-04-17:
Thank you. I tried adding this by hand, but no luck.

```
[root@home-assistant:/home/homeassistant/.homeassistant/.storage#](mailto:root@home-assistant:/home/homeassistant/.homeassistant/.storage#)  diff -Naur core.config_entries-178 core.config_entries
--- core.config_entries-178     2023-04-16 06:58:10.744025482 +0000
+++ core.config_entries 2023-04-16 07:03:53.056587843 +0000
@@ -250,7 +250,23 @@
                 "source": "ignore",
                 "disabled_by": null
+            },
+            {
+                "entry_id": "abcdefab41295b7164df4f04ca51dc7f",
+                "version": 1,
+                "domain": "apsystems_ecur",
+                "title": "ECU",
+                "data": {
+                        "host": "192.168.168.130",
+                        "scan_interval": 400
+                },
+                "options": {},
+                "pref_disable_new_entities": false,
+                "pref_disable_polling": false,
+                "source": "ignore",
+                "unique_id": null,
+                "disabled_by": null
             }
         ]
     }
}
```

If I compare the apsystems_ecur code to this:

https://developers.home-assistant.io/docs/creating_platform_index

especially:

https://github.com/home-assistant/example-custom-config/blob/master/custom_components/example_light/light.py

I think a function like the setup_platform() is missing in apsystems_ecur.

When I tried adding some years ago a beckhoff/ twincat component to home_assistant, there was a setup() function required.
I have to investigate howto do this nowadays with the modern async style.

Part/all of the setup_plattform() functionality is probably done by this HACS component if it is present. I have to check...

soundart, 2023-04-17:
I was not able to add a similarly structured example from the example repo either.

I have opened this:  https://github.com/home-assistant/example-custom-config/issues/43#issue-1671220141

soundart, 2023-04-18:
I managed now to start the integration. I used latest from git for home-assistant core and latest for apsystems_ecur.

So I guess  homeassistant 2022.5.5 is too old for latest apsystems_ecur.

closed 2023-04-19 by soundart