ksheumaker/homeassistant-apsystems_ecur #181: Frozen Integration with ECU-B several times a day

synack666, 2023-04-27:
Firmware: ECU_B_1.2.26A
Version apsystems_ecur: latest

Several times a day I get no more values, frozen at the last level. in this case Cache is activated.
I have as an example this morning the ECU rebooted, short improvement, after a restart of HA it goes again for an indefinite time. Online at APSystems the data are without gaps.

synack666, 2023-04-27:
[image]

m3tc0n, 2023-04-28:
I have the same problem. I have to restart the ECU (power cycle) to get it working again.

I thought it might have something to do with the strength of the zigbee signal, but it turns out that is not the case.

synack666, 2023-04-28:
From my point of view, the error message can be closed, since the OS 10.1 it runs without a single error, before several times a day.

m3tc0n, 2023-04-28:
I am also using OS 10.1, but I still have the problem. So for me, this can't be closed yet.

HAEdwin, 2023-04-28:
I can't solve this issue, what firmware versions are you running? HA doesn't need a restart, you should be fine with rebooting the ECU and switching the switch.ecu_query_device back on. Use the switch to be notified when the ECU needs a reboot or reboot the ECU automatically bij using a smart powerplug.

m3tc0n, 2023-05-01:
Sorry for the late response. I do not know the firmware version of the ECU at the moment, but I decided to buy a smart plug and automatically reboot the ECU the moment no more data is obtained.

I have the problem every day and to restart the ECU manually is very frustrating.

m3tc0n, 2023-05-03:
I checked the firmware. It is ECU_R_1.3.5A

m3tc0n, 2023-05-06:
This is typical for my ECU. Any solutions?

[image]

HAEdwin, 2023-05-07:
Might be a hardware issue, on which hardware are you running Home Assistant? It is known that Intel NUC's might have an issue.

m3tc0n, 2023-05-08:
Thanks for your reply. I'm running HA on a Raspberry Pi 4B, 4gb RAM, with a SSD.

HAEdwin, 2023-05-08:
Did you have a lot of trouble configuring the ECU or did you add/remove any panels? There might also be a configuration error. I remember that I encountered issues also after adding four panels. Ended up reconfiguring the whole PV and the issue was solved.

m3tc0n, 2023-05-08:
No, none of that. But if I read your post correctly, you recommend uninstalling and reinstalling the integration?

HAEdwin, 2023-05-08:
No that won't change the situation. I mean resetting the ECU to factory defaults and re-add the panels.

m3tc0n, 2023-05-08:
OK I will try that. 

For re-adding the panel, can I use the UID's of the inverters in de EMA app and put them in manually after resetting the ECU?

wemulder, 2023-05-30:
I've been having the issue with my ECU R stopping after a few days. It looks like the automation tries to reboot the ECU, but neither the automation or even the AP Systems app/website seem to be able to connect after that happens. To get it to connect again I need to physically reboot the ECU.
For the time being, I just put in a smart switch that reboots the ECU every morning at 5am to see if that resolves the issue. I also setup an automation to send me a notification if it's using cached data (if it happens during the day despite the daily reboot).
Looks like my firmware is: ECU_R_PRO_2.1.10.

cyberbit, 2023-05-30:
@wemulder I have the same problem as you, I use an automation to detect when cached data has been On for longer than usual, OR if the query device switch has been turned off, and run a script to reset the smart switch the ECU is using. I've been running this combo for several weeks without issue.

## Automation:
[image]
```yaml
alias: ECU Reset Watchdog
description: Call ECU Reset after long period ECU Using Cached Data being On
trigger:
  - platform: state
    entity_id:
      - binary_sensor.ecu_using_cached_data
    to: "on"
    for:
      hours: 0
      minutes: 16
      seconds: 0
  - platform: state
    entity_id:
      - switch.ecu_query_device
    to: "off"
    for:
      hours: 0
      minutes: 5
      seconds: 0
condition: []
action:
  - service: script.ecu_reset
    data: {}
mode: single
```

## Script:
[image]

```yaml
alias: ECU Reset
sequence:
  - type: turn_off
    device_id: a3xxxxxxxxxxxx
    entity_id: switch.ecu_breaker
    domain: switch
  - delay:
      hours: 0
      minutes: 0
      seconds: 8
      milliseconds: 0
  - type: turn_on
    device_id: a3xxxxxxxxxxxx
    entity_id: switch.ecu_breaker
    domain: switch
  - delay:
      hours: 0
      minutes: 0
      seconds: 30
      milliseconds: 0
  - type: turn_on
    device_id: 9fxxxxxxxxxxxx
    entity_id: switch.ecu_query_device
    domain: switch
mode: single
icon: mdi:refresh
```

wemulder, 2023-05-30:
@cyberbit Thanks for sharing that! If my daily reboot doesn't work I'll need to build something similar.
Will also continue to monitor firmware versions to see if a future update fixes this issue.

HAEdwin, 2023-10-29:
I'll close this issue because a workaround is in place. Might any other issue arrise feel free to open a new issue.

closed 2023-10-29 by HAEdwin