ksheumaker/homeassistant-apsystems_ecur #183: Wrong value

hekmanViii, 2023-04-29:
The sensor Ecu energy Today counts up the produceced energy, but once in a while is records the total energy of that day instead of the last xx miniutes. Which results in double the energy produced.

Ganzhans, 2023-05-31:
and is it possible to correct it by hand and ...... how ?

Ganzhans, 2023-05-31:
I see the [ECU today Energy] entry but I don't now what the correct values mus be .......
I edit some thing, but what must change only the Engery section or the live section also ?

hekmanViii, 2023-06-17:
[image]

hekmanViii, 2023-06-17:
See post before this happens about once or twice every week. The day total comes in place of the total for 19:40.

HAEdwin, 2023-11-15:
I am currently evaluating a change that uses the historical data that the ECU stores. This has a number of advantages, including a correct display of Energy Today in /100 resolution regardless of HA restart. There is a small difference between the Today Energy and EMA website. I always look at how big that difference is and how this can arise.

HAEdwin, 2023-11-19:
The pre-release v1.2.31b contains an extra ECU query that introduces (part of) the use of energy history data. This is a query I previously described but was never needed (we thought). To compensate load time for the integration, I shortened the sleep-time between opening and closing the port. I also introduced the keepalive, a careful introduction to removing this sleep-time and keeping the port open. Success will depend greatly on compatibility with different ECU models and firmware. If you own an older model starting with 2160xxxxxxx ECU-R, you can even leave out the open/close functions, but does this work for other ECU models? I could make it optional in the UI settings. Anyway it should solve this issue. Please let me know if it does so that I can close the issue.

HAEdwin, 2023-11-20:
@hekmanViii @Ganzhans 

If you are trying v1.2.31b you might have an issue with the Today Energy. I have forgotten to specify END in the suffix.

Line 38 of APSsystemsSocket.py should be changed from `self.ecu_energy_history_suffix = "END00\n"` to `self.ecu_energy_history_suffix = "END00END\n"`. Then restart HA and see if that made a change to Today's Energy.

HAEdwin, 2023-11-21:
I'm pretty sure Today Energy's issue has been resolved for the ECU-R, ECU-B and ECU-C. I'm closing this issue, if anything pops up, feel free to create a new issue. Notice that there might still be a small difference between the ECU en EMA values. EMA does it's own calculation but is sometimes adjusted by the maintenance interaction with the ECU in the evening.

closed 2023-11-21 by HAEdwin