ksheumaker/homeassistant-apsystems_ecur #185: Any experience with multiple QT2 invertors via ECU-R?

schalkk, 2023-05-01:
Hi,

Has anybody tried to retrieve data from the QT2 (3 phase / 4 Channel Micro invertor) via the ECU-R?  When I do this, the system only recognizes the first of the 6 active invertors.
In the invertor_qty it does show "6".

Any pointers would be appreciated.

schalkk, 2023-05-04:
I now have all the inverters working after switching to HA via the HACS installation. Sees all 6 the inverters.

However, two issues remain:
1. Sees the inverters as YC1000 although it is a QT2 (all 6 starts with 901)
2. The daily energy and lifetime energy does not match with what I see in the APS mobile App or on the EMA website. HA is always lower.

HAEdwin, 2023-05-07:
New model inverter that should be compatible with the QS1 but isn't yet identified as one. There are two steps in solving this.
1. Please enable debug logging: [Settings]>[Devices & Services]>choose the menu to 
[image]
2. wait for the next interval to pass and stop the debug logging. A log file will be downloaded. Examine this log and look for the lines where the [custom_components.apsystems_ecur.APSystemsSocket] is outputting (less meaningfull) data. I need the second data string of the three available. Please post it here.
3. I will add the ID to the integration after verification.

schalkk, 2023-05-08:
Hi,
here you go!
03 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313931303030323030303100062023050813190090100000420801303201f4008b00cc019c00d2019f00d6019900cd90100000576401303201f4008b00d3019d00d6019c00e5019900db90100000698301303201f4008900cc019a00d6019e00d7019a000090100000623401303201f4008c0097019700d3019900e6019f00e490100000694601303201f4008d00cb019800d6019b00e0019c00df90100000735201303201f4008b00bf019800e6019500f1019700ec454e440a'

schalkk, 2023-05-08:
I see in HA that the Timzone for the ECU is set to GMT-8 while in the EMA App, it is showing Europe/Amsterdam.  What could be causing this?

I stil have the question as to why the Total Daily Energy and Lifetime Energy do not match between HA and ECU app or EMA Website.  HA is always behind.  Could the timezone be impacting this?

schalkk, 2023-05-08:
Hi,
Thanks for the quick update! Much appreciated,

For me the difference between the number is not small.
- Lifetime - HA 854 vs EMA 1160
- Daily - HA 13.74 vs EMA 19.88
Is this data retrieved from the ECU or is it aggregated in HA?

HAEdwin, 2023-05-08:
What interval is set for the integration? I would recommend setting it to 300. But I might have to dive deeper into this. Fact is that the daily values should not differ to much.

HAEdwin, 2023-05-08:
Lifetime energy shown in HA is data taken from the ECU. This same data is sent to EMA but is not being used so I assume that the EMA site uses different calculations.

schalkk, 2023-05-08:
For interest sake, I just connected to the AP on the ECU with EME MGR to check the values on the ECU-R.

It seems that HA and EMA MGR lines up but EMA App (and Website) is way above.

**Lifetime Energy**

- HA - 856.5
- EMA Mgr (AP) - 856.44
- EMA app (internet) - 1,160

**Daily Energy**

- HA - 15.72
- EMA MGR - 15 
- EMA App - 21.4

Really Strange!  Which numbers should we trust?

schalkk, 2023-05-08:
I have asked APS support for clarification.

HAEdwin, 2023-05-13:
@schalkk Did you happen to receive a reply from APsystems yet?

schalkk, 2023-05-15:
@HAEdwin  No reply yet.  Sending them another e-mail.  This time specifically to EMEA Support.

schalkk, 2023-05-23:
@HAEdwin 

Still no reply.

I have watched the ECU Current Power and Today Energy very closely.  I compared the numbers in the EMA Web with the numbers in Home Assistant.

It seems that the problem is related to the "Today Energy" (same with Lifetime Energy) starting to increment, every day, hours after the Current Power.  The Current Power starts incrementing shortly after sunrise, but the Today Energy number only start to show it's first non-zero number around 10:30 to 11:00 every mornning.

Here is today's example:
[image]

Once "Today Energy" starts to increment, it stays in step with the number shown in the EMA website.

Here is the pattern for the last 3 days:
[image]

Just an observation.

HAEdwin, 2023-05-23:
Not sure why it is increasing late in your graphs, below are mine. Did you check the time zone and correct time on devices?
[image]
[image]

schalkk, 2023-05-24:
@HAEdwin Really strange!  I have check using the EMA Mgr app, it shows the correct time and timezone.  If I check via the port 8899 I get GMT-8, which is not correct.
[image]

Is the one API call for GetRealTimeData (code 0002) collecting all the values or do you collect energy via the GetEnergyOfWeekMonthYear (code 0040)? Could they be using different timestamps (as seen by the EMA Mgr app & API)

schalkk, 2023-09-07:
@HAEdwin This issue has now been sent to the APSystems programmers according to the Tech Support team.

HAEdwin, 2023-10-29:
I suppose there was no answer from APsystems? Daily energy, timezone as well as a missing MAC address for the wireless interface are all ECU firmware bugs that have been around for a long time, no effort is being done to solve it. 

For the daily energy; it is not being sent to EMA (I sniffed the data), it is calculated at EMA. I have to think of a way to calculate kWH within the integration. See also https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/207

YC1000 and QT2 share the same protocol. In the new release you'll also see the QT2 mentioned. 

The difference between the lifetime energy in HA versus EMA is strange. Lifetime energy from the ECU is being sent to EMA but I guess they wanted to synchronize it with the actual data being sent to EMA and is therefor calculated at EMA. Thing is that there is a five minute interval which means that there is always a estimate... a cloud can shortly appear in front of the sun that is not picked up bij the ECU and vice versa.  The push to EMA and the pull for HA are not synchronized so there always will be a difference.

HAEdwin, 2023-11-16:
I am currently evaluating the values ​​when using the history data read by the ECU. The daily energy value is a better match than the previously read value, so I will implement it in a beta version.

HAEdwin, 2023-11-19:
The pre-release v1.2.31b contains an extra ECU query that introduces (part of) the use of energy history data. This is a query I previously described but was never needed (we thought). To compensate load time for the integration, I shortened the sleep-time between opening and closing the port. I also introduced the keepalive, a careful introduction to removing this sleep-time and keeping the port open. Success will depend greatly on compatibility with different ECU models and firmware. If you own an older model starting with 2160xxxxxxx ECU-R, you can even leave out the open/close functions, but does this work for other ECU models? I could make it optional in the UI settings. Anyway it should solve this issue. Please let me know if it does so that I can close the issue.

schalkk, 2023-11-20:
Hi,
I installed the new beta, restarted HA.
The results look much the same to me:
From: HA 
[image]
From APSystemsema.com
[image]

On HA the daily energy is still 0 @11:51am and the Lifetime Energy is way lower than EMA (6.28MWh vs. 4,357.9kWh)

Thanks for the effort!

HAEdwin, 2023-11-20:
I noticed that EMA itself does the calculation for Today's Energy and Life Time Energy. You are on firmware version 1.3.6C with the ECU 2160xxxxx (that's the older model ECU-R without SunSpec logo on the back right?) - I'm still on firmware 1.2.25b with the same type of ECU so there's a difference also.

If you enable debug logging for the integration, what does the log show? I am especially interested in the four lines of the log after "Querying ECU...".

I might have forgotten to specify END in the suffix. 
Line 38 of APSsystemsSocket.py should then be changed from self.ecu_energy_history_suffix = "END00\n" to `self.ecu_energy_history_suffix = "END00END\n"`. Then restart HA and see if that made a change to Today's Energy.

schalkk, 2023-11-20:
Hi,
Yes, the ECU does not have the SunSpec Logo.  I received it in March this year.
Daily energy is now starting to report numbers after the code change.

Debug info:
2023-11-20 15:23:22.422 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Querying ECU...
2023-11-20 15:23:25.429 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303039343030303132313630303031323038333030310000aa3d0000028e00000011d0d0d0d0d0d0d00006000631303031324543555f525f312e332e36433030394574632f474d542d3880971b05ce12000000000000454e440a'
2023-11-20 15:23:34.451 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313931303030323030303100062023112015193490100000735201303201f3007b001a0186001a0187001b017f001990100000694601303201f30078001a0182001b0186001d0185001d90100000623401303201f30079001c0180001d0185001e0187001e90100000698301303201f3007900180186001d0183001c0189001c90100000576401303201f30076001b0187001a0182001e0183001d90100000420801303201f30077001a0188001a018600160182001b454e440a'
2023-11-20 15:23:34.451 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303630303033303030901000007352ab901000006946ab901000006234ab901000006983ab901000005764ab901000004208ab454e440a'
2023-11-20 15:23:34.451 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'41505331323030373630303034303030302023111400000000202311150000000020231116000000032023111700000005202311180000000020231119000000002023112000000011454e440a'
2023-11-20 15:23:34.451 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Got data from ECU
2023-11-20 15:23:34.452 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Returning {'timestamp': '2023-11-20 15:19:34', 'inverters': {'901000007352': {'uid': '901000007352', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 23, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [26, 26, 27, 25], 'voltage': [390, 391, 383]}, '901000006946': {'uid': '901000006946', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 20, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [26, 27, 29, 29], 'voltage': [386, 390, 389]}, '901000006234': {'uid': '901000006234', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 21, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [28, 29, 30, 30], 'voltage': [384, 389, 391]}, '901000006983': {'uid': '901000006983', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 21, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [24, 29, 28, 28], 'voltage': [390, 387, 393]}, '901000005764': {'uid': '901000005764', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 18, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [27, 26, 30, 29], 'voltage': [391, 386, 387]}, '901000004208': {'uid': '901000004208', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 19, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [26, 26, 22, 27], 'voltage': [392, 390, 386]}}, 'today_energy': 0.17, 'ecu_id': '216000120830', 'lifetime_energy': 4358.1, 'current_power': 654, 'qty_of_inverters': 6, 'qty_of_online_inverters': 6, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}
2023-11-20 15:23:34.453 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 12.032 seconds (success: True)

HAEdwin, 2023-11-20:
Thanks for testing, this looks much better now. Don't forget to disable debug logging again ;)

schalkk, 2023-11-21:
Sorry, today the Today's Energy and Lifetime Energy is not updating at all.
[image]
 Debug Log:
 2023-11-21 10:07:20.525 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Querying ECU...
2023-11-21 10:07:23.533 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303039343030303132313630303031323038333030310000aa410000003f00000000d0d0d0d0d0d0d00006000131303031324543555f525f312e332e36433030394574632f474d542d3880971b05ce12000000000000454e440a'
2023-11-21 10:07:32.553 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313931303030323030303100062023112110035490100000735200303200000064000001890000018a001b0188001a90100000694601303201f30074000f01880010018b001001850010901000006234003032000000640000018200000188000801850008901000006983003032000000640000018300000189000401860004901000005764003032000000640000018400000189000801810008901000004208003032000000640000018400000186000401890004454e440a'
2023-11-21 10:07:32.553 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'41505331313030363030303330303090100000735200901000006946a890100000623400901000006983009010000057640090100000420800454e440a'
2023-11-21 10:07:32.553 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'41505331323030373630303034303030302023111500000000202311160000000320231117000000052023111800000000202311190000000020231120000000342023112100000000454e440a'
2023-11-21 10:07:32.553 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Got data from ECU
2023-11-21 10:07:32.556 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Returning {'timestamp': '2023-11-21 10:03:54', 'inverters': {'901000007352': {'uid': '901000007352', 'online': False, 'signal': 0, 'frequency': 0.0, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [0, 0, 27, 26], 'voltage': [393, 394, 392]}, '901000006946': {'uid': '901000006946', 'online': True, 'signal': 65, 'frequency': 49.9, 'temperature': 16, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [15, 16, 16, 16], 'voltage': [392, 395, 389]}, '901000006234': {'uid': '901000006234', 'online': False, 'signal': 0, 'frequency': 0.0, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [0, 0, 8, 8], 'voltage': [386, 392, 389]}, '901000006983': {'uid': '901000006983', 'online': False, 'signal': 0, 'frequency': 0.0, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [0, 0, 4, 4], 'voltage': [387, 393, 390]}, '901000005764': {'uid': '901000005764', 'online': False, 'signal': 0, 'frequency': 0.0, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [0, 0, 8, 8], 'voltage': [388, 393, 385]}, '901000004208': {'uid': '901000004208', 'online': False, 'signal': 0, 'frequency': 0.0, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [0, 0, 4, 4], 'voltage': [388, 390, 393]}}, 'today_energy': 0.0, 'ecu_id': '216000120830', 'lifetime_energy': 4358.5, 'current_power': 63, 'qty_of_inverters': 6, 'qty_of_online_inverters': 1, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}
2023-11-21 10:07:32.556 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 12.034 seconds (success: True)

Later in the afternoon:
2023-11-21 15:19:20.422 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Querying ECU...
2023-11-21 15:19:23.430 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303039343030303132313630303031323038333030310000aa410000018e00000000d0d0d0d0d0d0d00006000631303031324543555f525f312e332e36433030394574632f474d542d3880971b05ce12000000000000454e440a'
2023-11-21 15:19:32.452 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313931303030323030303100062023112115135490100000735201303201f30075001401850014018a00140189001390100000694601303201f30073001201880013018900140188001490100000623401303201f400740013018b0014018700150186001590100000698301303201f40076001201870015018b00130187001390100000576401303201f400720013018a0012018a00150184001490100000420801303201f30073001101860012018e001101880012454e440a'
2023-11-21 15:19:32.452 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303630303033303030901000007352aa901000006946aa901000006234a9901000006983aa901000005764a9901000004208a8454e440a'
2023-11-21 15:19:32.452 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'41505331323030373630303034303030302023111500000000202311160000000320231117000000052023111800000000202311190000000020231120000000342023112100000000454e440a'
2023-11-21 15:19:32.453 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Got data from ECU
2023-11-21 15:19:32.453 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Returning {'timestamp': '2023-11-21 15:13:54', 'inverters': {'901000007352': {'uid': '901000007352', 'online': True, 'signal': 66, 'frequency': 49.9, 'temperature': 17, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [20, 20, 20, 19], 'voltage': [389, 394, 393]}, '901000006946': {'uid': '901000006946', 'online': True, 'signal': 66, 'frequency': 49.9, 'temperature': 15, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [18, 19, 20, 20], 'voltage': [392, 393, 392]}, '901000006234': {'uid': '901000006234', 'online': True, 'signal': 66, 'frequency': 50.0, 'temperature': 16, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [19, 20, 21, 21], 'voltage': [395, 391, 390]}, '901000006983': {'uid': '901000006983', 'online': True, 'signal': 66, 'frequency': 50.0, 'temperature': 18, 'model': [... cut ...] 901000004208': {'uid': '901000004208', 'online': True, 'signal': 65, 'frequency': 49.9, 'temperature': 15, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [17, 18, 17, 18], 'voltage': [390, 398, 392]}}, 'today_energy': 0.0, 'ecu_id': '216000120830', 'lifetime_energy': 4358.5, 'current_power': 398, 'qty_of_inverters': 6, 'qty_of_online_inverters': 6, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}

I do set the ECU Query Device off at sunset and the set it on again at sunrise.

HAEdwin, 2023-11-22:
@schalkk Strange, I don't think this is due to the integration right now. All the values in the debug log are set to zero so that's what is being read and displayed.  The Life Time value of zero means that the ECU must have reset somehow or might have a defect. EMA calculates the Today Energy and Life Time Energy seperate from the ECU so these numbers don't always match up with HA. There are 18 people that have updated to the latest release until now. Have you powercycled the ECU allready?

HAEdwin, 2023-11-22:
On the 21th of november your ECU status was:
ECU ID: 216000xxxxxx
ECU model:1
Lifetime Energy: 4358,5 kWh
Current Power: 63 Watt
Today Energy: 0 kWh
Last EMA connect:D0D0D0D0D0D0D0
Inverters: 1 of 6 online
Signal Channel:10
Firmware Version:ECU_R_1.3.6C

At 10:07 in the morning, but still only one inverter online. 
----------------  Inverter data  ----------------
Date/time:20231121100354
ECU model:01
----------------
Inverter ID:901000007352
Inverter online:0
Inverter mode:YC1000
No Data / Offline
----------------
Inverter ID:901000006946
Inverter online:1
Inverter mode:YC1000
Inverter Frequency:49,9 Hz
Inverter Temperature:16 Degrees
Inverter Power Ch1:15 W
Inverter Voltage Ch1:392 Volts
Inverter Power Ch2:16 W
Inverter Voltage Ch2:395 Volts
Inverter Power Ch3:16 W
Inverter Voltage Ch3:389 Volts
Inverter Power Ch4:16 W
----------------
Inverter ID:901000006234
Inverter online:0
Inverter mode:YC1000
No Data / Offline
----------------
Inverter ID:901000006983
Inverter online:0
Inverter mode:YC1000
No Data / Offline
----------------
Inverter ID:901000005764
Inverter online:0
Inverter mode:YC1000
No Data / Offline
----------------
Inverter ID:901000004208
Inverter online:0
Inverter mode:YC1000
No Data / Offline

How are the panels positioned?

schalkk, 2023-11-22:
Hi,

Here are the latest debug after a restart of the ECU-R.

2023-11-22 13:34:11.193 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Querying ECU...
2023-11-22 13:34:14.204 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303039343030303132313630303031323038333030310000aa41000006db00000000d0d0d0d0d0d0d00006000631303031324543555f525f312e332e36433030394574632f474d542d3880971b05ce12000000000000454e440a'
2023-11-22 13:34:23.225 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313931303030323030303100062023112213331590100000735201303201f30075004301920044018d00450192004490100000694601303201f40074003e01900047018f004d0191004d90100000623401303201f40075004301930045018b00470191004790100000698301303201f400740057018f00480193004e0190004b90100000576401303201f40072004b0191004d0193004f018d004b90100000420801303201f30073004f01960047018f005001900046454e440a'
2023-11-22 13:34:23.225 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303630303033303030901000007352ad901000006946ad901000006234ad901000006983ad901000005764ad901000004208ad454e440a'
2023-11-22 13:34:23.225 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'41505331323030373630303034303030302023111600000003202311170000000520231118000000002023111900000000202311200000003420231121000000002023112200000000454e440a'
2023-11-22 13:34:23.225 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Got data from ECU
2023-11-22 13:34:23.226 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Returning {'timestamp': '2023-11-22 13:33:15', 'inverters': {'901000007352': {'uid': '901000007352', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 17, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [67, 68, 69, 68], 'voltage': [402, 397, 402]}, '901000006946': {'uid': '901000006946', 'online': True, 'signal': 67, 'frequency': 50.0, 'temperature': 16, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [62, 71, 77, 77], 'voltage': [400, 399, 401]}, '901000006234': {'uid': '901000006234', 'online': True, 'signal': 67, 'frequency': 50.0, 'temperature': 17, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [67, 69, 71, 71], 'voltage': [403, 395, 401]}, '901000006983': {'uid': '901000006983', 'online': True, 'signal': 67, 'frequency': 50.0, 'temperature': 16, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [87, 72, 78, 75], 'voltage': [399, 403, 400]}, '901000005764': {'uid': '901000005764', 'online': True, 'signal': 67, 'frequency': 50.0, 'temperature': 14, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [75, 77, 79, 75], 'voltage': [401, 403, 397]}, '901000004208': {'uid': '901000004208', 'online': True, 'signal': 67, 'frequency': 49.9, 'temperature': 15, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [79, 71, 80, 70], 'voltage': [406, 399, 400]}}, 'today_energy': 0.0, 'ecu_id': '216000120830', 'lifetime_energy': 4358.5, 'current_power': 1755, 'qty_of_inverters': 6, 'qty_of_online_inverters': 6, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}

HAEdwin, 2023-11-22:
Got some zero values today also for Today Energy, weird...
[image]

schalkk, 2023-11-22:
Upgraded to 1.2.31 and restarted HA.  No change.

HAEdwin, 2023-11-22:
What if you install the previous version? v1.2.30

HAEdwin, 2023-11-22:
I suppose approach is like:

- kWh = the sum of 12 power measurements in Watts / 12 (max interval is 5 minutes so you should have 12 measurements to get to one hour) / 1000  
- Today Energy in kWh = the total of each sum from the previous line within a day

Life time Energy: LTE = LTE + DTE

schalkk, 2023-11-23:
Same problem with the previous version:
[image]
Result:
[image]

schalkk, 2023-11-23:
On the 22nd of August I had a different firmware version: Aug 1.3.6**A** Nov 1.3.6**C**
[image]
Does anybody know when 1.3.6C was released?

HAEdwin, 2023-11-25:
@schalkk I think the best option will be to calculate Today Energy and LifeTime Energy within the integration. It's best to remove the option to set the interval, the max update frequency is preferred anyway so I will set that to 5 minutes. This way it will be easy to calculate kWh. The only disadvantage is that people will initially have to equate their Lifetime Energy to the EMA site.

HAEdwin, 2024-01-11:
@schalkk Currently I've added three extra sensors on DEV where I calculate the LifeTime Energy, Daily Energy and (new sensor) Hourly Energy myself to compare them to the internal values from the ECU. This will also resolves the occasional zero values.

I've allredy seen measurement deviations that have to do with rounding. I can imagine that this will paint a more positive picture on the EMA site over a longer period of time. I will continue the measurements. In the mean time firmware versions on the ECU have been updated so I keep that in mind. 
[image]

HAEdwin, 2024-01-21:
@schalkk There have been several firmware updates in the mean time. 
Is Today Energy Still Behind Current Power? 
I have used my own calculations to see what it yielded, but in the end it remains an approximation and yields little. I notice a few things; 
1. There is rounding, both from the ECU (first rounding) and within the EMA (second rounding). 
2. The query interval is not synchronized with the ECU updates. Together with the sometimes failing queries, this can lead to a deviation in the measurement results.

A solution could be to run the first ECU query more often and look at the timestamp of the last ECU update. If it differs from the previous one, run a full query.

arnovannijnatten, 2024-01-22:
For me it is still about 5-7 kWh off on a daily basis. Anything generated under 5kWh does not even get registered and reports as 0 for Today energy.
This is also the case for using the EMA Manager app and getting the Today energy data from there. It is very much less than what the consumer EMA app reports. The EMA consumer app is actually much closer to what my electricity net meter registers.
So having a (self) calculated energy today would make sense for my situation.
I am on firmware ECU_R_1.3.9 now.

HAEdwin, 2024-01-22:
@arnovannijnatten I think there is something wrong with your ECU, try to replace it under warranty if possible.

HAEdwin, 2025-03-17:
I will close this issue, feel free to open a new discussion on my git at https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader if needed.