ksheumaker/homeassistant-apsystems_ecur #187: Stop working after HA update

lbarreiro, 2023-05-05:
It stop working after the last update with this msg in log:

Este erro teve origem numa integração personalizada.

Logger: homeassistant.config_entries
Source: custom_components/apsystems_ecur/__init__.py:200 
Integration: APSystems PV solar ECU (documentation) 
First occurred: 07:36:57 (1 occurrences) 
Last logged: 07:36:57

Error setting up entry ECU: 216000061340 for apsystems_ecur
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/apsystems_ecur/__init__.py", line 200, in async_setup_entry
    hass.config_entries.async_setup_platforms(config, PLATFORMS)
AttributeError: 'ConfigEntries' object has no attribute 'async_setup_platforms'

renini, 2023-05-05:
You probably have to upgrade the custom component. Probably simular as #186

lbarreiro, 2023-05-05:
Yeap... same didn't have the latest version... sorry
Issue Close and solved

closed 2023-05-05 by lbarreiro