ksheumaker/homeassistant-apsystems_ecur #188: connection refused

TimWinkler, 2023-05-05:
This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/apsystems_ecur/__init__.py:199
Integration: APSystems PV solar ECU (documentation, issues)
First occurred: 12:52:12 PM (1 occurrences)
Last logged: 12:52:12 PM

Error setting up entry ECU: 216300036646 for apsystems_ecur
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/apsystems_ecur/__init__.py", line 199, in async_setup_entry
    hass.config_entries.async_setup_platforms(config, PLATFORMS)
AttributeError: 'ConfigEntries' object has no attribute 'async_setup_platforms'

renini, 2023-05-05:
Seems like you are using an old version of the custom component. Same as #186 and #187

TimWinkler, 2023-05-05:
i think i am on v1.2.29

TimWinkler, 2023-05-05:
restore to 2023.4.6 fixed it - broken with 2023.5.1

Henttonen, 2023-05-16:
Is there anything to be done? 
it's a shame this has worked well until now, the latest Home Assistant Core update came again and it didn't start working =(

Henttonen, 2023-05-30:
Hi,
Is there any easy way to fix this?

Logger: homeassistant.config_entries
Source: custom_components/apsystems_ecur/__init__.py:187
Integration: APSystems PV solar ECU ([documentation](https://github.com/ksheumaker/homeassistant-apsystems_ecur), [issues](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues))
First occurred: 12.48.53 (2 occurrences)
Last logged: 13.22.52

Error setting up entry ECU: XxXxXxXxXxXxXx for apsystems_ecur
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/apsystems_ecur/__init__.py", line 187, in async_setup_entry
    hass.config_entries.async_setup_platforms(config, PLATFORMS)
AttributeError: 'ConfigEntries' object has no attribute 'async_setup_platforms'

HAEdwin, 2023-05-30:
What version of the integration are you running on?

HAEdwin, 2023-06-14:
I'll close this issue since the original question for this issue was solved.

closed 2023-06-14 by HAEdwin