ksheumaker/homeassistant-apsystems_ecur #190: Unable to connect adter update Home Assistant OS 10.2

OscarDev-, 2023-05-31:
Hi,

Yesterday I updated my Home Assistant OS to version 10.2 and since I updated that, the APS systemts integration is unable to connect.

The ECU-C is still accessible trough its web interface with my browser as well as trough a terminal in HomeAssistant at its IP-address

I have the following message in the log:
2023-05-31 07:19:18.260 WARNING (SyncWorker_1) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: ECU returned 0 for lifetime energy, this is either a glitch from the ECU or a brand new installed ECU. Raw Data=b''

Might there be some incompatibility with the latest update to HA OS 10.2? 
I am running the latest version of the integration (1.2.9)

HAEdwin, 2023-05-31:
If you take a look at the web interface, what is the value of the lifetime energy? I can't relate this to the HA OS 10.2 update for this is not an error message but the integration simply reads 0 for lifetime energy.

OscarDev-, 2023-06-01:
It has the current values:
[image]

OscarDev-, 2023-06-01:
You can close this issue. I did an power cycle on the ECU-C and it seems to work again. Weird that it happened after updating to HA OS 10.2 tho.
Thank you for checking into it.

closed 2023-06-01 by OscarDev-