ksheumaker/homeassistant-apsystems_ecur #191: No data from ECU-R (2160x) - only solution power cycle

jan-dewit, 2023-06-05:
I am using a ECU-R (starting with 2160) with HA 2023.5.4 and integration v1.2.29

At random times the integration stops displaying data and is frozen, while the APS EMA app still shows all the data.

Only solution here is to power cycle the ECU-R, after which it immediately works again.
So it seems the ECU-R is still forwarding data to the cloud but the ECU-R is not responding on the LAN anymore.
I did test the connectivity on port 8899 in a frozen situation, and it gives connection refused.

Would love to provide logging or more information to solve this issue.

jan-dewit, 2023-06-05:
> This is a known phenomenon but interesting to know which ECU firmware version you have. I have the same ECU and am still on version 1.2.24 Below you can see the frequency at which it is not possible to query the ECU in my case. If, according to my setting, that would happen 5 times in a row, the integration would give up and the ECU query device switch would switch off so that you could receive a signal that you have to powercycle the ECU. But this happens once every two months or so. [image]
> 
> what might be going on here is port exhaustion of some kind, I checked the code but everything is being opened and closed correctly.

Thanks for your reply. Im on firmware: ECU_R_1.2.24B
At your side it seems only not reachable for a short period of time, here it stays unreachable:
[image]

HAEdwin, 2023-06-05:
It's also an option to stop querying the ECU after sunset by setting the ECU query Device switch to the off position via an automation (and turn it back on at sunrise), this might lenghten the periode for which the ECU stays "alive". In the integration settings look for "retries when ECU fails" increase the value to 8 or so to see if that helps. You can also send a notify to your phone when the ECU query Device switch turn to the off position.

jan-dewit, 2023-06-14:
> the ECU after sunset by setting the ECU query Device switch to the off position via an automation (and turn it back on at sunrise), this might lenghten the periode for which the ECU stays "alive". In the integration settings look for "retries when ECU fails" increase the

thanks for the tips! Luckily it did not fail anymore since more than a week, so its really unpredictable :)

HAEdwin, 2023-06-14:
ECU-R should be stable just like the ECU-B unless changes in the 1.2.24b firmware broke something. I'm still at 1.2.24. I'll close this issue, might any question pop-up feel free to do so.

closed 2023-06-14 by HAEdwin