ksheumaker/homeassistant-apsystems_ecur #192: [No Data] Script crash

max3163, 2023-06-06:
Since this morning, the query script crash with this error : 

```
Traceback (most recent call last):
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/ECU_B.py", line 48, in <module>
    data = loop.run_until_complete(ecu.async_query_ecu())
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/APSystemsECU.py", line 124, in async_query_ecu
    self.ecu_raw_data = await self.async_send_read_from_socket(cmd)
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/APSystemsECU.py", line 101, in async_send_read_from_socket
    return await asyncio.wait_for(self.async_read_from_socket(), timeout=self.timeout)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
    return fut.result()
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/APSystemsECU.py", line 93, in async_read_from_socket
    raise APSystemsInvalidData(error)
APSystemsECU.APSystemsInvalidData: End suffix (b'END\n') missing from ECU response end_data=b'1\x00\x00\n' data=b'APS110095000121630005826801\x00\x00\n'
```

When I change the end pattern to match with the new format, the script just crash a few lines after ... 

``` 
Traceback (most recent call last):
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/ECU_B.py", line 48, in <module>
    data = loop.run_until_complete(ecu.async_query_ecu())
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/APSystemsECU.py", line 126, in async_query_ecu
    self.process_ecu_data()
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/APSystemsECU.py", line 234, in process_ecu_data
    self.check_ecu_checksum(data, "ECU Query")
  File "/k8s_storage/domoticz-config-pvc-43977a86-d8db-4206-94e9-fce6d4179d97/scripts/python/ECU/APSystemsECU.py", line 211, in check_ecu_checksum
    raise APSystemsInvalidData(error)
APSystemsECU.APSystemsInvalidData: Checksum on 'ECU Query' failed checksum=95 datalen=29 data=b'41505331313030393530303031323136333030303538323638303100000a'
``` 

I guess an update change the format of the protocol and the query is not correct now ... 

How can I help you to fix this ??

max3163, 2023-06-07:
I've already powercycle my ECU-B without sucess....
One person on the domoticz forum have had this issue few days ago and it come back OK without doing anything. 
For me, so far it's still KO...

HAEdwin, 2023-06-07:
Is all the data received malformed? If not, you could temporary ignore these datastrings and wait until some days are passed.

max3163, 2023-06-08:
Sadly all queries failed... and it's still not working ... 
There are some users with the same problems on the domoticz forum but no solution so far...

HAEdwin, 2023-06-09:
All contributions are welcome

HAEdwin, 2023-06-11:
All contributions are welcome but I think the spinoff focusses on Domoticz and this git is all about HA. To avoid further confusion I'll close this issue since it is not part of an existing issue for this integration or its code.

closed 2023-06-11 by HAEdwin