ksheumaker/homeassistant-apsystems_ecur #207: ECU R with QT2 inverters with today energy 0.0

arnovannijnatten, 2023-10-28:
I have an ECU R with 6 QT2 inverters. When I started a couple of weeks ago with the integration, all seemed fine. Although HA consistently reported less kWh produced than the APS EMA app and energy returned to the grid by my smart meter (Dutch). It was off by around 5kWh on a day 45 kWh energy was reported by the smart meter.

Now however, I am not getting any readings on most days of the week. Out of the past 7 days, 4 consistently returned todays energy 0.0.

I have today's debug log attached below. The app says I produced 5.7 kWh, the integration shows 0.0.

```
2023-10-28 17:40:22.274 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur] Querying ECU...
2023-10-28 17:40:27.597 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303039343030303132313630303031323438373630310000a7fc0000006e00000000d0d0d0d0d0d0d00006000631303031324543555f525f312e332e36413030394574632f474d542d3880971b05ac90000000000000454e440a'
2023-10-28 17:40:38.138 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313931303030323030303100062023102817383090100000111701303201f30075000401840004018600040183000490100000303201303201f30075000401850003018600040184000590100000387201303201f30075000401870003018300030184000490100000276101303201f30078000501830005018800050182000590100000337201303201f30079000601830005018900060181000690100000395401303201f300770005018400040188000601840006454e440a'
2023-10-28 17:40:38.138 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303630303033303030901000001117d1901000003032d3901000003872c4901000002761d4901000003372c8901000003954c0454e440a'
2023-10-28 17:40:38.139 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur] Got data from ECU
2023-10-28 17:40:38.139 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur] Returning {'model': 'YC1000', 'channel_qty': 4, 'power': [5, 4, 6, 6], 'voltage': [388, 392, 388], 'inverters': {'901000001117': {'uid': '901000001117', 'online': True, 'signal': 81, 'frequency': 49.9, 'temperature': 17, 'model': 'YC1000', 'channel_qty': 4, 'power': [4, 4, 4, 4], 'voltage': [388, 390, 387]}, '901000003032': {'uid': '901000003032', 'online': True, 'signal': 82, 'frequency': 49.9, 'temperature': 17, 'model': 'YC1000', 'channel_qty': 4, 'power': [4, 3, 4, 5], 'voltage': [389, 390, 388]}, '901000003872': {'uid': '901000003872', 'online': True, 'signal': 76, 'frequency': 49.9, 'temperature': 17, 'model': 'YC1000', 'channel_qty': 4, 'power': [4, 3, 3, 4], 'voltage': [391, 387, 388]}, '901000002761': {'uid': '901000002761', 'online': True, 'signal': 83, 'frequency': 49.9, 'temperature': 20, 'model': 'YC1000', 'channel_qty': 4, 'power': [5, 5, 5, 5], 'voltage': [387, 392, 386]}, '901000003372': {'uid': '901000003372', 'online': True, 'signal': 78, 'frequency': 49.9, 'temperature': 21, 'model': 'YC1000', 'channel_qty': 4, 'power': [6, 5, 6, 6], 'voltage': [387, 393, 385]}, '901000003954': {'uid': '901000003954', 'online': True, 'signal': 75, 'frequency': 49.9, 'temperature': 19, 'model': 'YC1000', 'channel_qty': 4, 'power': [5, 4, 6, 6], 'voltage': [388, 392, 388]}}, 'ecu_id': '216000124876', 'today_energy': 0.0, 'lifetime_energy': 4300.4, 'current_power': 110, 'qty_of_inverters': 6, 'qty_of_online_inverters': 6, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}
```

I'm using:
Home Assistant 2023.10.5
Frontend 20231005.0 - latest
Integration 1.2.29
ECU firmware 1.3.6A, connected via WLAN

HAEdwin, 2023-10-28:
In the data there is no sign of 5.7 kWh but indeed 0 kWh. This is as you mention the difference of approximately 5 kWh per day. I have no idea where this could come from, it is not a bug with this HA integration, it just reads the value from the ECU. Do you also have a graph for today's energy in kWh? I'm thinking of a configuration error in the (time) zone or an incorrect reference time from another host. The value should increase immediately after sunrise (if there is sunshine).

arnovannijnatten, 2023-10-28:
Thanks for the quick reply. 

The graph looks like this:
[image]

Also the time zone setting is configured correctky I believe.
[image]

HAEdwin, 2023-10-28:
Sorry, I mean the HA graph

arnovannijnatten, 2023-10-28:
That shows 0 for the whole day. 
[image]

While current power did actually record readings. 
[image]

HAEdwin, 2023-10-28:
I think the today energy is calculated on the EMA server based on the uploaded data because the ECU itself does not keep track of this properly for all firmware versions (which it does for my ECU of course). I will have to find a solution for this by having the value calculated by the integration itself.

HAEdwin, 2023-11-19:
The pre-release v1.2.31b contains an extra ECU query that introduces (part of) the use of energy history data. This is a query I previously described but was never needed (we thought). To compensate load time for the integration, I shortened the sleep-time between opening and closing the port. I also introduced the keepalive, a careful introduction to removing this sleep-time and keeping the port open. Success will depend greatly on compatibility with different ECU models and firmware. If you own an older model starting with 2160xxxxxxx ECU-R, you can even leave out the open/close functions, but does this work for other ECU models? I could make it optional in the UI settings. Anyway it should solve this issue. Please let me know if it does so that I can close the issue.

arnovannijnatten, 2023-11-19:
Thanks for the update! I've upgraded to this beta version and will keep.you posted

HAEdwin, 2023-11-21:
I'm pretty sure Today Energy's issue has been resolved for the ECU-R, ECU-B and ECU-C. I'm closing this issue, if anything pops up, feel free to create a new issue.

arnovannijnatten, 2023-11-22:
I noticed you reopened #185 
Same here: no noticable change after upgrade to the latest version of the integration. All inverters are online, but daily energy is 0.
[image]

HAEdwin, 2023-11-22:
@arnovannijnatten Tell me about your ECU type and firmware please.

arnovannijnatten, 2023-11-22:
> I have an ECU R with 6 QT2 inverters. 
...
 ECU firmware 1.3.6A, connected via WLAN

By now firmware has been upgraded to 1.3.6C

HAEdwin, 2023-11-22:
I updated https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/185 Bad news about this new firmware... I'll close this issue so we can continue there.

closed 2023-11-24 by HAEdwin