ksheumaker/homeassistant-apsystems_ecur #209: could not extract checksum int from 'Energy History Query' data=b''

albatorsk, 2023-11-20:
After updating the integration to version 1.2.31b, it fails to work. Downgrading to 1.2.30 resolves the problem and it starts working again.

With version 1.2.31b, this line comes up multiple times in the HA log:
2023-11-20 10:43:32.381 WARNING (SyncWorker_11) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: could not extract checksum int from 'Energy History Query' data=b''

I have an ECU-R Pro with firmware 2.1.15. I tried restarting HA as well as the ECU-R.

Let me know if there's anything I can do to help resolve this.

HAEdwin, 2023-11-20:
Thanks for your feedback! Do you own an ECU starting with ID 2160xxxx? That's without the SunSpec logo on the back of the ECU? Please read this https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/185#issuecomment-1818979180 hopefully it solves the issue.

albatorsk, 2023-11-20:
Not exactly 2160xxxx but it starts with 2162xxxx. It does have a SunSpec logo on the underside of it.

I tried your suggestion anyway of modifying APSsystemsSocket.py and that helped. I am now using v1.2.31b and the integration works again.

HAEdwin, 2023-11-20:
Thanks for testing!

wildekek, 2023-11-21:
Same issue here on a ECU-C 2150xxxx

HAEdwin, 2023-11-21:
Did you allready apply the #185 fix?

wildekek, 2023-11-21:
> I have forgotten to specify END in the suffix. Line 38 of APSsystemsSocket.py should then be changed from `self.ecu_energy_history_suffix = "END00\n"` to `self.ecu_energy_history_suffix = "END00END\n"`. Then restart HA and see if that made a change to Today's Energy.

Yup, that fixes the issue. Thanks a bunch!

HAEdwin, 2023-11-21:
Thank you for testing!

closed 2023-11-21 by HAEdwin