ksheumaker/homeassistant-apsystems_ecur #210: ECU-R Lifetime and Today energy not updating

schalkk, 2023-11-21:
What I am seeing today after the upgrade to the latest Beta:
[image]

**Debug Log:**
2023-11-21 15:19:20.422 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Querying ECU...
2023-11-21 15:19:23.430 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303039343030303132313630303031323038333030310000aa410000018e00000000d0d0d0d0d0d0d00006000631303031324543555f525f312e332e36433030394574632f474d542d3880971b05ce12000000000000454e440a'
2023-11-21 15:19:32.452 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313931303030323030303100062023112115135490100000735201303201f30075001401850014018a00140189001390100000694601303201f30073001201880013018900140188001490100000623401303201f400740013018b0014018700150186001590100000698301303201f40076001201870015018b00130187001390100000576401303201f400720013018a0012018a00150184001490100000420801303201f30073001101860012018e001101880012454e440a'
2023-11-21 15:19:32.452 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303630303033303030901000007352aa901000006946aa901000006234a9901000006983aa901000005764a9901000004208a8454e440a'
2023-11-21 15:19:32.452 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur.APSystemsSocket] b'41505331323030373630303034303030302023111500000000202311160000000320231117000000052023111800000000202311190000000020231120000000342023112100000000454e440a'
2023-11-21 15:19:32.453 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Got data from ECU
2023-11-21 15:19:32.453 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Returning {'timestamp': '2023-11-21 15:13:54', 'inverters': {'901000007352': {'uid': '901000007352', 'online': True, 'signal': 66, 'frequency': 49.9, 'temperature': 17, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [20, 20, 20, 19], 'voltage': [389, 394, 393]}, '901000006946': {'uid': '901000006946', 'online': True, 'signal': 66, 'frequency': 49.9, 'temperature': 15, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [18, 19, 20, 20], 'voltage': [392, 393, 392]}, '901000006234': {'uid': '901000006234', 'online': True, 'signal': 66, 'frequency': 50.0, 'temperature': 16, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [19, 20, 21, 21], 'voltage': [395, 391, 390]}, '901000006983': {'uid': '901000006983', 'online': True, 'signal': 66, 'frequency': 50.0, 'temperature': 18, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [18, 21, 19, 19], 'voltage': [391, 395, 391]}, '901000005764': {'uid': '901000005764', 'online': True, 'signal': 66, 'frequency': 50.0, 'temperature': 14, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [19, 18, 21, 20], 'voltage': [394, 394, 388]}, '901000004208': {'uid': '901000004208', 'online': True, 'signal': 65, 'frequency': 49.9, 'temperature': 15, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [17, 18, 17, 18], 'voltage': [390, 398, 392]}}, 'today_energy': 0.0, 'ecu_id': '216000120830', 'lifetime_energy': 4358.5, 'current_power': 398, 'qty_of_inverters': 6, 'qty_of_online_inverters': 6, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}

HAEdwin, 2023-11-21:
Did you allready restart Home Assistant after installation? History value for the 21st of this month is indeed zero. The Life Time Energy has not changed in any way for this new version.

schalkk, 2023-11-22:
Yes, I did restart it.  I also restarted it again this morning, but I am seeing the same results.

[image]

HAEdwin, 2023-11-22:
I reopened #185

closed 2023-11-22 by HAEdwin