ksheumaker/homeassistant-apsystems_ecur #211: Removal of inverter after replacement

blafois, 2023-11-21:
Hello!

Thanks for your great Home Assistant integration.

Unfortunately, one of my QT2 inverter died, and I replaced it using the "replace" feature of EMA app.
The new inverter was detected after a reload/restart of HA.
However, the old one is still there, and I did not find any option to remove it. I could only disable it.

Would be nice to detect removed inverters and delete them, or provide an option to do so !

Thanks for the great job !

HAEdwin, 2023-11-21:
Thank you, it is much appreciated! I'd say it's more a HA thing. If the device is not linked to anything anymore, then you can delete it. So make sure that nothing creates the device again from outside HA, the ECU must have completely removed the device (restart ECU). Remember to also remove all associated entities (also filtered ones). Then go to the Devices & Services section, click on the Devices tab, select the specific device you want to remove, in the Device Info card you'll find three dots at the bottom, click on it and choose Delete.

c0mplex1, 2023-11-28:
Hi,

I'm in the same situation.
One of my YC600 inverters is dead, and is replaced by a DS3.
The new inverter was detected after a restart of HA, but the old one is still there.
I have checked that all associated entities were removed, but at the bottom of the Device Info card I can't find the three dots to delete the old inverter.
Did I forget something?
Please advise.
[image]

blafois, 2023-11-28:
Same here, have followed @HAEdwin  recommendations, without success. Unlinked all entites. Still not able to delete the device. Had to manually remove entries in `core.device_registry` file... but I don't like that !

c0mplex1, 2023-11-29:
Thanks to @blafois's answer, I successfully removed the inverter by manually removing it from the core.device_registry file.

HAEdwin, 2023-12-02:
@blafois @c0mplex1 The integration never opt in to allow the user to delete a device from the UI, this has to be implemented because I agree with you that this is now missing. I'll have to find time to fix this.

HAEdwin, 2023-12-17:
@blafois @c0mplex1 Added the opt in to remove a device in v1.2.30b. I have successfully removed inverters and checked the core.device_registry file for correct changes. In my test cases the inverter returns after HA restart because it is still configured in the ECU.
[image]

c0mplex1, 2023-12-17:
Thanks for the fix and the update.
I don't have a defective inverter at the moment, but I will test it next time I have to replace another defective inverter.

blafois, 2023-12-17:
Great work @HAEdwin ! Thank you very much for the hard work and the great binding ! I have manually removed from EMA manager an inverter to confirm I'm able to then delete it in HA. 
We can close this request

HAEdwin, 2023-12-21:
Thank you! I'll close this issue.

closed 2023-12-21 by HAEdwin