ksheumaker/homeassistant-apsystems_ecur #212: ECU do not respond as expected

grandyoz, 2023-11-22:
Hello,

I have received my ECU-R today and I have connected it to my local network. The ECU-R updated its firware.
I have assigned a fixed IP and I can ping it.
But, when I try the command "nc -v 172.16.0.4 8899" with my local ECU-R IP, it respond nc: connectx to 192.168.0.xx port 8899 (tcp) failed: Connection refused.

Can you help me to identify the problem ?

Thanks in advance !

grandyoz, 2023-11-22:
I am also wired connected. I think we do a mistake and shall be connected wireless If I read well the doc. Does it change something ?
I have another problem to connect it wireless... but it's not the place do discuss about that ! @mcattle, did you succeed to connect it wireless ? If yes, can we talk on another place ? Perhaps you can help me !
Bye

HAEdwin, 2023-11-22:
@grandyoz Things might change with other firmware releases, but best is to follow the documentation - if things have changed feel free to update the documentatie (PR) or notify me. Thing is that I'm currently the only maintainer and cannot test all ECU models for example. Ethernet port 8899 = closed  Wireless port 8899 = open

grandyoz, 2023-11-22:
@HAEdwin , yes, if I found something, it would be a pleasure to make a PR.
I am in the same case of @mcattle ... only the port 4540 is open. But I am in wired connection.
@HAEdwin , do you know, for your case, if the connection type is important ? Moreover, if you run nc -z -n -v 10.0.0.30 1-65535 2>&1 | grep succeeded, could you tell us what are the ports opened ?

My ECU is a 216000 series and it's on 1.3.6C version.

Thanks

HAEdwin, 2023-11-22:
Yes, we've investigated the option to use 4540 3 years ago but gave up on that. For that moment 8899 was more succesful. ECU-R pro needs wireless connection to function with this integration. See documentation.

grandyoz, 2023-11-22:
ok, thank you VERY much for that. So I have to find my connection problem (wireless) before to continue.
I will tell you soon !
Thanks

HAEdwin, 2023-11-22:
Because all ECU's are being setup with the ECUapp the 8899 method seems more durable and uses the same method. 

I totally forgot I made an 4540 listener in the past but we never got to decoding the data as good as we did for the 8899 method.
Port 4540 outputs data with an 5 minute interval and is only available when you have a ethernet connected ECU.  Can't remember how much data was sent on that port.

mcattle, 2023-11-23:
That's kind of a shame the 4540 listener isn't functional, as it would mean one less device on my WiFi.  :-)

@grandyoz Here's what I did to switch the network connection over:

1. Install the EMA Manager App (not the EMA app for end users) on my phone.
2. Unplug the Ethernet and restart the ECU.
3. Connect my phone to the ECU's WiFi hotspot (ECU_R_2160xxxxxxxx) and use 88888888 as the password (Eight 8's)
    a. If you don't see the hotspot right away, press the AP button on the left side to put it in AP mode and it should show up eventually.
4. Open the EMA Manager App, and click `Local Access` below the login button to connect to the ECU locally.
5. In the app, press the `Workspace` tab in the bottom menu.
6. Open `ECU network settings`, choose `WLAN`, and enable it, then choose your network SSID and enter the password.
7. Check that the ECU is connected to your WiFi network.
8. In my case, it connected to the WiFi, but could not connect to the cloud server (grey bullet on the main page that should be green, and second light on the front of the unit did not light up).  I simply unplugged the unit, then plugged it back in, and it was fully connected after it booted back up.
9. Finally, do the port 8899 test.  I got the response I was looking for on my second try, and have successfully added it to Home Assistant.

I didn't see a way to set the WLAN IP address on the unit (maybe set the LAN IP address manually when WLAN is enabled?), but at any rate I reserved a fixed IP address for the unit on my router's DHCP server, so it'll always be assigned the same address.

closed 2023-11-23 by grandyoz