ksheumaker/homeassistant-apsystems_ecur #213: 1.30.0 breaks integration

marcusds, 2023-11-22:
After updating to `1.30.0` today I am now getting the error:
`Using cached data from last successful communication from ECU. Invalid data error: ECU returned 0 for lifetime energy, this is either a glitch from the ECU or a brand new installed ECU. Raw Data=b''`

I tried downgrading to `1.29.0` and am getting the same error, though I wonder if that is just the downgrade not working.

There is nothing else interesting in the logs, so I am not sure what other info to give.

`Firmware: ECU_R_PRO_2.1.15`

coldenvy, 2023-11-23:
same here ran into the same issue thanks for confirming

HAEdwin, 2023-11-24:
I removed version 1.30.0 right now but kept the beta version. Is you see an error in the log please put the integration into debug mode. That way I'm better able to see what's going wrong. This issue but also the firmware issue with firmware 1.3.6C will keep me busy for a while :( 
As allways, things are working fine for my good old ECU-R.

Robsonator, 2023-11-25:
Same problem for me.
In Debug Mode i got the following message:
2023-11-25 16:33:11.263 WARNING (SyncWorker_1) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: ECU returned 0 for lifetime energy, this is either a glitch from the ECU or a brand new installed ECU. Raw Data=b''

niouniou49, 2023-11-27:
same here with 1.2.30 the communication to ECU-R brakes after few minutes; rolled back to .29 and will see...
some ideas

HAEdwin, 2023-11-27:
Yesterday I released an updated version of 1.2.30 with fewer changes left in it like the keepalive. If you want to give this version a try? Please provide enough info (debug log, firmware version). Thank you for testing!

niouniou49, 2023-11-28:
same problem continues....
after few hours with the last release, it breaks and i need to restart the ECU electrically :-)
i roll backed to .29 for fews hours and will let u know
my ecu -r is from 2020 and start with 2160xxxxx

CarlosGS, 2023-12-01:
Same problem, here's some debugging info:
- **After downgrading to 1.2.29 and physically restarting the ECU, the next day it is working again.**
- We noticed that the testing command `nc -v 192.168.x.x 8899` used to give connection refused, but now it works. It is an ECU-B and the reply is `0013ECU_B_1.2.27B009Etc/GMT-8` in case the version helps.
- Also, in Home Assistant the logs for the ECU plugin said "connection refused"

However, it did stay connected to the internet, and it still sent data to the official APsystems sever/app.
**This points to the ECU entering some odd state where it just can't accept connections in port 8899.**

Maybe the new code is leaving TCP connections open, or doing something different that could be messing with the ECU's TCP server? :thinking: 

Here's the diff in case someone can give it a try https://github.com/ksheumaker/homeassistant-apsystems_ecur/compare/v1.2.29...v1.2.30#files_bucket
At first I only see there's a new keepalive flag for the socket, maybe that could fill up the ECU's TCP clients queue.

HAEdwin, 2023-12-01:
Thanks @CarlosGS you might be on the right track. If you are able to modify the file like so, you might have ruled out the error. I'm not able to reproduce the error, every ECU model/firmware seems to act differently
[image]

niouniou49, 2023-12-01:
i commented the line with keep alive as in your example and will let you know if it works correctly after that

TheRealCryss, 2023-12-01:
Commenting out that line didn't work, for me at least. I get a 'host unreachable' message when trying the terminal test connect with 'nc'. The ip is correct and I see it in my router.

HAEdwin, 2023-12-01:
@TheRealCryss Have you just installed the integration (new user) or allready a user? Please add the type of ECU and firmware version.

TheRealCryss, 2023-12-01:
Hm. I did. I'll try again.

TheRealCryss, 2023-12-01:
Update: I got the integration up again (without switching the ECU on and off again).

The reason was poor or not good enough Wifi connectivity it seems. I checked my Fritzbox router and saw that 3 of the 4 Tasmota WiFi plugs also weren't connected to the network and that the 2.4Ghz spectrum was busy between 90 and 100%!
The ECU had a good downstream of about 36Mbit but only 1MBit upstream. So I guess that was the reason why I couldn't test the connection with 'nc' even while the ECU was connected to my Wifi.

Now I must see if the connection remains stable.

niouniou49, 2023-12-02:
1.2.30 even with patch line 88... failed
back to 1.2.29 for today. 
ECU-R-EU 216xxxx  version ECU_R_1.2.25B
can i update the ECU? new firmware version?

CarlosGS, 2023-12-02:
Thanks for giving it a try! Unfortunately [our setup](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/213#issuecomment-1835659120) rolling back to version 1.2.29, though it worked yesterday, didn't work today. It shows cached data since 2AM.

Testing the `nc` command again shows Connection refused. I've physically restarted the ECU and it hasn't changed.
Still, the ECU is connected to the internet (both lights ON) and the APSystem app works. Strange.

HAEdwin, 2023-12-02:
@CarlosGS Just to make sure, did you assign a fixed IP-address to the ECU? Could it be that the DHCP server assigned a new IP-address making it unreachable on the old address?

niouniou49, 2023-12-03:
after i rolled back to 1.2.29 - all is working back again ... after 26h, no wifi network interruption, all data went on with the sun coming back. i can ping the fixed ip and the result of the command 'nc -v 10.xxxxxx 8899' is 'Connection to 10.xxxxxx port 8899 [tcp/*] succeeded!
made a script to switch off the ecu queries at 6pm and switch on at 8:30 am, then data came back .
there is something to investigate in the 1.2.30 release that broke the integration withECU-R-EU 216xxxx version.

CarlosGS, 2023-12-03:
@HAEdwin Yep, same IP. Ping works, but refuses connection on port 8899. I've rebooted HASS & the ECU and still does it.

@niouniou49 In case wifi affects this, is your ECU-R connected by wifi, or by cable? Mine is ECU-B (wifi).

CarlosGS, 2023-12-06:
This is our picture so far with 1.2.29. Maybe it was a coincidence when we downgraded ~nov 26th :confused:
[image]
(Gray=Working well, Yellow=Cached, Dark=No data)

@niouniou49 If it says ECU-R-EU then it would be ECU-R? Our label shows ECU-B-EU 216xxxx, also connected via wifi.

HAEdwin, 2023-12-23:
In v1.2.30b I was a bit conservative with the changes by rolling them back. It teaches me to release carefully and to take sufficient time for experiences from the user group. For now I close this issue. Feel free to start a new issue based on your experiences with the last 1.2.30 release. (v1.2.30 is the official release of v1.2.30b for the non-beta testers among the users)

closed 2023-12-23 by HAEdwin

marcusds, 2023-12-23:
So far it seems to be working fine.