ksheumaker/homeassistant-apsystems_ecur #214: When inverters are off, inverter sensors entities should not update

wildekek, 2023-11-29:
When the inverters are off, the following sensors report a value of 0:
- sensor.inverter_xxxxxxx_power_ch_1
- sensor.inverter_xxxxxxx_power_ch_2
- sensor.inverter_xxxxxxx_frequency
- sensor.inverter_xxxxxxx_voltage

The following sensors report their last know state:
- sensor.inverter_xxxxxxx_signal

The correct behavior is not updating the state, like the temperature sensor:
- sensor.inverter_xxxxxxx_temperature

Wrong:
[image]

Expected behavior:
[image]

HAEdwin, 2023-11-29:
The temperature sensor(s) of the inverter(s) also reported zero when switching off the inverter(s). This zero value does not reflect the actual temperature of the inverter. That is why the preference arose not to update these graphs.

wildekek, 2023-11-30:
Yes, that makes total sense. So you agree this behavior should apply to the other graphs as well?

HAEdwin, 2023-12-02:
When we would do this, frequency sensor graphs would show up like a series of lines __   __   ___   ___   ___
I created a poll to see how others think about this idea: https://github.com/ksheumaker/homeassistant-apsystems_ecur/discussions/215

HAEdwin, 2023-12-21:
I'll try to make it optional like so:
[image]

wildekek, 2023-12-21:
Sweet, that we would be a great addition!

HAEdwin, 2023-12-23:
I'm currently testing... If you would like to test, you can find the sources here: https://github.com/HAEdwin/homeassistant-apsystems_ecur. 

Personally I like to update the graphs when the inverters are offline because this morning it showed power being zero on two channels with the inverter online, showing that these parameters are indeed zero and not a guess.
[image]

HAEdwin, 2023-12-26:
@wildekek ik ben klaar met testen. De v1.3.0 beta staat klaar voor installatie en testen :tada:

HAEdwin, 2023-12-28:
@wildekek With some delay due to the busy Christmas holidays and a software bug the 1.3.0b version includes the feature - ready for further testing 🎉

wildekek, 2023-12-28:
Nice work @HAEdwin ! I've installed it and will check out the results in 24h. Thanks a bunch and happy holidays to you.

HAEdwin, 2024-01-01:
Fixed: Caught a bug on a fresh install; when configuring the integration it throws an error on the no_graphs because there was no default so the key and value (False) could not be created.

Release was updated just now. @wildekek, All is in working as expected?

wildekek, 2024-01-03:
Yep, thanks for your work!

closed 2024-01-03 by wildekek