ksheumaker/homeassistant-apsystems_ecur #219: 2 ecu-b together not working

patmtp35, 2023-12-28:
strange behavior, I have two ecu-b, when I include one it works correctly, if I include the second one, it brings up the dataq and the first does not bring it up anymore... if I override the first and then re -included, it is the second which no longer reports the datas.
i have a ping on etch others et the ping it's ok

HAEdwin, 2023-12-28:
From the readme instructions: It currently supports one ECU (one instance of the integration) but a work-around is available https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/142 Hope this works for you.

patmtp35, 2023-12-28:
ok tanks for the prompt answer , i ll try to do this solution 
regards

closed 2023-12-28 by patmtp35