ksheumaker/homeassistant-apsystems_ecur #220: Cant find ECU or  life-time energy is 0

balla1337, 2024-01-06:
Won't work any longer i noticed today, can,t connect. Checked connection,but App/ECU all is fine and IP still correct.

HAEdwin, 2024-01-06:
What are the results of the https://github.com/ksheumaker/homeassistant-apsystems_ecur#test-your-connection ? If that part works fine then the life-time energy must be zero for some reason. Did you install a new ECU? Have you noticed a firmware update? What type of ECU do you have? Also use the search function and read this: https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/198

balla1337, 2024-01-07:
Thank you for fast reply man, i have ECU-B hard-reset + repeater reset recently fixed my Problem. Have a nice day

closed 2024-01-07 by balla1337

HAEdwin, 2024-01-07:
Thanks for the feedback on your own solution to reset the ECU-B and repeater. 👍