ksheumaker/homeassistant-apsystems_ecur #222: Not updating data

dkwolf1, 2024-01-09:
I have a problem with updating data. I've try to install every version. The query doesn't work well. It's not updating automatically. Several times a day I have to switch the query button manually and even then it's not updating every time. I'm using the WiFi ecu-r variant. First half of 2023 it al worked fine...

HAEdwin, 2024-01-12:
How often do you query the ECU? Is it being queried by one instance of HA? Try a powercycle. You could use an automation to enable the query button again. At times you encounter the non reachable ECU, are you able to ping it? Does it help when you position the ECU somewhere else in the room? Did you check the zigbee signal data?

smidtd, 2024-01-26:
I have the same problem. ECU-R 2160. When I reinstalled the integration it updated a few times but after some time I get the following error:
This error originated from a custom integration.

Logger: custom_components.apsystems_ecur
Source: custom_components/apsystems_ecur/__init__.py:58
Integration: APSystems PV solar ECU (documentation, issues)
First occurred: 22 January 2024 at 13:10:24 (2 occurrences)
Last logged: 14:37:01

Communication with the ECU failed after 5 repeated attempts.

I used the integration for a year without any problems. Is there a fix for this?

dkwolf1, 2024-01-26:
I have running an automation that set the query on every hour. But it's stil strange..

HAEdwin, 2024-01-28:
@smidtd It is difficult to anticipate the changing status of different ECUs and their firmware updates. If your ECU has a webui despite the fact that it starts with ID 2160, you can try to see if the software reset works. On line 62 of __init__.py you will find `if (self.cached_data.get("ecu_id", None)[0:3] == "215") or (self.cached_data.get("ecu_id", None)[0 :4] == "2162"):` This line checks which type of ECU can apply to the software reset. If you change 2162 at the end of the line to 2160 (and HA restart), the reset procedure is used. This means the ECU will not suffer less from outages, but the ECU will reboot automatically.

smidtd, 2024-01-30:
@HAEdwin : I have access to the Ecu WiFi (ECU_R_###). How can I test if my ECU has a webui? I tried to connect to it's local IP (10.10.100.254) but no reply. Does it need a certain port to connect to?
I did get the ECU Manager app to work. In the app was no option for a software reset.

HAEdwin, 2024-01-31:
http://x.x.x.x/index.php/hidden 
If you are able to reach this page you have an ECU-R pro. What ECU firmware version is installed? Was there a change in you local network configuration (firewall, DHCP)? Do not block the EMA url's, this cause the ECU to stop working.

smidtd, 2024-02-01:
I don't have the webui so no ECU-R pro. ECU firmware is ECU_R_1.2.27.
No changes in network config (firewall or DHCP).
The ECU Query just stops somewhere during the day. When I switch it back on manually everything works again.
I will build an automation that will trigger the switch after "Device off".
I really like this HACS integration so it would be perfect if another solution would be found.
Thanks @HAEdwin for your replies!

1 February 2024
ECU Query Device turned on triggered by service Switch: Turn on
09:26:18 - 1 hour ago - DdS
31 January 2024
ECU Query Device turned off
17:13:44 - 17 hours ago
ECU Query Device turned on triggered by service Switch: Turn on
16:43:25 - 18 hours ago - DdS
ECU Query Device turned off
14:37:39 - 20 hours ago
ECU Query Device turned on triggered by service Switch: Turn on
12:25:56 - 22 hours ago - DdS

HAEdwin, 2024-02-01:
No problem @smidtd 
I have the same ECU-R and firmware. It is normal if you are unable to execute the query on the ECU once in a while. After all, the ECU also does IO with the EMA site every 5 minutes and due to timing changes they can meet each other. The integration anticipates this by accessing the cache and trying again in five minutes. However, if it does not work (consecutively) after the set number of attempts, the ECU Query switch flips so that an automation can respond to this by, for example, controlling a smart plug and power cycling the ECU. The placement of the ECU is important, sometimes it helps to move the ECU.

From the start we encountered failing queries and we managed to limit them as much as possible. Each attempt such as re-query after failure led to greater unavailability. Ultimately, using the cache turned out to be the best option and anticipates on all ECU models.

My ECU is placed in the middle between the two small PV panelarrays." It's not very common that I get query errors. Closest accesspoint is 5 meters away, inverters are at 8 meters away from the ECU.

[image]

smidtd, 2024-02-02:
Thanks for the explanation @HAEdwin. That gives a lot more insight.
I've now added an automation to switch back on the query. Let's see what that does for the next days/weeks.
Have a great weekend!

dkwolf1, 2024-03-08:
I checked today and with an automation it's working!

closed 2024-03-20 by HAEdwin