ksheumaker/homeassistant-apsystems_ecur #226: ECU-B update and production yesterday

freeman59, 2024-01-29:
Hello, I have been having a problem for some time with the integration and my ECU-B. 

I see that it takes a lot of time before we have data on the big day, whereas before it was more regular. 
In addition, this indicates erroneous production. Example yesterday and the day before yesterday, the production of my panels on the APSystem site were 2.34 and 2.24kWh, on the HA side I only have 1.31 and 1.19kWh!

Do you know what is the problem?

Regards

HAEdwin, 2024-01-29:
Hi @freeman59 If you mean ECU life, you could be right. I perform a simultaneous calculation and in a few days the value is already higher than that of the ECU. There is nothing that can be done about this at the moment. If I have time I will post an update where you can optionally choose the calculated values.

freeman59, 2024-01-29:
thanks for your reply! What do you mean by ECU Life? 
I hadn't observed a gap when I started using the plugin on HA. I have the impression that this problem has been happening since a few updates. It's possible?

HAEdwin, 2024-01-29:
That's ECU life time Energy. It is read and generated from the ECU itself. The integration only indicates for this entity what the ECU itself has registered. I am running simultaneous calculations and these values are allready higher so I suppose they will reflect the values on the EMA website much better over time. Here's an example (production entities are calculated values):
[image]

freeman59, 2024-01-29:
[image]

On my side, you can see that on APS website, my current ECU Lifetime production is 174.65kWh and where yesterday it was 2.24kWh

On HA side, the problem seems to be the Lifetime totally different (since weeks), cause it says that the ECU lifetime is only at 79.2kWh (missing near 100kWh)

[image]

and for yesterday : 
[image]

HAEdwin, 2024-01-29:
If the ECU has not been reset to factory defaults in the meantime, the difference is indeed far too great. This is not due to the integration.

freeman59, 2024-01-30:
Well I don't know what's happen, I don't know how to reset factory but I never reset the ECU since I own it (it was installed on October 2023). Do you know how to have the good value returned to HA ? I mean, the actual value is false under HA not on the ECU so, is it possible to restore ? Do I need to uninstall the package and reinstall it ? Do you think it will work ? I don't really care about history from now, I just need to have the good value each day :)

thanks !

HAEdwin, 2024-01-31:
Allthough the ECU's lifetime energy is being sent to the EMA website, the EMA website does not reflect the ECU's registration of lifetime energy, it does it's own calculation. The lifetime energy that is displayed in HA is just being read from the ECU and cannot be corrected, the ECU contains the wrong value, this is not due to the package (integration). You won't loose history by reinstalling the package but it won't solve the difference between the value the ECU holds and the EMA website. You might want to contact your vendor or APSystems about the large difference in values between your ECU and the EMA site.

freeman59, 2024-02-02:
Thanks but I don't understand what you mean 

the plugin doesn't work with my ECU, so bad. Lifetime and also daily value are incorrect ... I think I will add a Shelly 3EM to get information from my APS QT2

HAEdwin, 2024-02-02:
> thanks for your reply! What do you mean by ECU Life? I hadn't observed a gap when I started using the plugin on HA. I have the impression that this problem has been happening since a few updates. It's possible?

No this is not possible, there was no change in the way the integration calls the value for the ECU's Lifetime Energy. There could be an issue with a firmware update on your ECU that might have happened in the mean time. This is not an Integration issue. But indeed the best way to measure is a direct (smart)meter on the grid.

Is it allright to close this issue?

freeman59, 2024-02-10:
Well this plugin will not be useful for me :( I will switch to a Shelly 3EM to recover production of my APS

thanks anyway

HAEdwin, 2024-02-10:
You should check your ECU-B, hopefully with a next firmware update your issue will be solved.

closed 2024-02-10 by HAEdwin