ksheumaker/homeassistant-apsystems_ecur #227: Unknown error occurred

giagrand74, 2024-01-29:
Unable to complete the setup
Already tried to restart the ECU-B and restart HA
Device is available on the network and reply correctly to nc -v 172.168.1.56 8899
v 1.3.1

Core
2024.1.5
Supervisor
2023.12.1
Operating System
11.4
Frontend
20240104.0
[image]

HAEdwin, 2024-01-29:
What if you take an older version of the integration 1.2.30 for example?

HAEdwin, 2024-01-31:
Thanks for the feedback, there might be a bug somewhere, I'll have to look into that. Was it a complete new install?

giagrand74, 2024-02-01:
Update to version 1.3.1
All seems ok till now.
I just downgrade to 1.2.30 and updated again today, I did not removed the integration from HACS.

HAEdwin, 2024-02-08:
Strange, not sure what the issue was because I was not able to reproduce it. Might any other question arrise, feel free to open a new issue.

closed 2024-02-08 by HAEdwin