ksheumaker/homeassistant-apsystems_ecur #230: Wrong data since last update

christianvaes, 2024-02-12:
Hi, I'm using the APSystems ECU-R integration version 1.4.0 running on home assistant:

My details
ECU 2160
Firmware: ECU_R_1.3.10
8 DS3-L invertors
Core 2024.2.1
Supervisor 2024.01.1
Operating System 11.5
Frontend 20240207.1
Home assistant is runing as a VM on my Qnap nas.

Since, I think this update(not totally sure), the separate inverters show 0W power pretty often. Sometimes one inverter sometimes all of them. I also feel the current power is way off because of this.

There are no records in the log and earlier everything worked fine. I already restarted home assistant, the ECU R and all inverters but this only makes a difference for a short time.

What can I do to fix this? The values in official the EMA App/web UI  from Apsystems are showing up fine.

HAEdwin, 2024-02-12:
I don't think this is due to the update, my 2160 is at firmware ECU_R_1.2.27 so I'll keep an eye on that. Receiving 0 watts is strange when the EMA site doesn't show this, though I know EMA does it's own calculations and the ECU will validate values against EMA later in the evening when the inverters are down. If the integration receives 0 this can't be misinterpreted unless  other data is also showing strange values. Might also be the firmware... Also try to position the ECU elsewhere just to see if that helps.

christianvaes, 2024-02-12:
Thank you for your reply. I will keep an eye on this hopefully it will solve itself as it appeared, it was just that it seems to be at the same moment the update of the integration went live. But it can also be a firmware update. I didn't keep an eye on when the ecu R updated as yours got an older firmware.

christianvaes, 2024-02-13:
Hi, today I have issues again. Now it seems to not report 0W but all entities are "stuck" at other values. In the logging it now shows an error:

```
Logger: custom_components.apsystems_ecur
Source: custom_components/apsystems_ecur/__init__.py:135
Integration: APSystems PV solar ECU ([documentation](https://github.com/ksheumaker/homeassistant-apsystems_ecur), [issues](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues))
First occurred: 10:25:46 (4 occurrences)
Last logged: 10:46:46

Using cached data from last successful communication from ECU. Invalid data error: Result on 'Inverter data' incorrect end signature '202' != END data=b'4150533131303031353030303230320a'
```

After this no results in home assistant anymore.

But when I run:

```
nc -v 10.0.0.14 8899
Connection to 10.0.0.14 port 8899 [tcp/*] succeeded!
APS1100160001END
APS110094000121600009696701��K	������10012ECU_R_1.3.10009Etc/GMT-8��`�END
```

It seems to work fine, also the EMA platform of APsystems is updated fine.

HAEdwin, 2024-02-13:
Weird, seems like other data is put through the port. Can you enable debug logging and post them form me? Have you allready powercycled the ECU? 
[image]

christianvaes, 2024-02-13:
I have enabled debug logging now and power-cycled the device. Will keep you posted.

christianvaes, 2024-02-14:
Today two of the invertors showed the 0W values again(time: 10:35) when they were working fine on the apsystems app. I think it will be in this debug log, another thing I noticed is that the "download diagnostics" button for this integration throws an error(also in this log available, time 10:42). @HAEdwin  you check my log please?

[ha_log_14022024.txt](https://github.com/ksheumaker/homeassistant-apsystems_ecur/files/14278175/ha_log_14022024.txt)

HAEdwin, 2024-02-14:
It appears that the protocol has not changed, data is read from the ECU and displayed accordingly. The integration only reads what is in the ECU in terms of data at that moment. If the APSystems app (is that the EMA app?) shows different than it might be still showing old data, ignoring the issue or somewhat else like lagging real time issues and might fall into error when the issue continues for a longer time period. From what I see in the log, the integration does function ok, it does not read or interpret data wrong - try to determen how you can really tell that they are down.

christianvaes, 2024-02-14:
Thank you, this is really helpful, I didn't think about the other parameters from the integration. It seems the frequency is rock solid between 49,9 and 50hz. The voltage also seems to be fine between 230 and 235. However the signal seems to have an issue since 4th of februari. Before that date there was never a signal drop(only some fluctuations) however after that date it seems that randomly the connection with one or multiple investors break down. Seems something is wrong with the ecu or  interference as you describe. Will try to check if I can do anything about that. Did you see such issue before?

HAEdwin, 2024-02-14:
Yes, some even remove the WiFi antenna from the ECU.

christianvaes, 2024-02-14:
Would it help to connect the ecu wired instead of using wifi and will the integration still work? Based on the readme I thought that wouldn't work for this integration and the version of my ecu.

HAEdwin, 2024-02-14:
No that will not work, on the 2160 series the port cannot be queried when the ECU is ethernet connected.

HAEdwin, 2024-02-14:
APSystems did the right move with their EZ- inverter series.

christianvaes, 2024-02-14:
Yes if wifi would work better. I positioned my antennas the same as you did now let's see if it will help tomorrow.
Thank you for all the help.

RkcCorian, 2024-02-15:
Hi together, I can confirm that since the 4th Februar e.g. the values ```ecu_current_power``` and ```ecu_today_energy``` at minimum are not correct. As I have ioBroker still in place and there this is still working fine I have to say that this is a bug in the Home Assistant integration. There was an update of the integration that I have taken, not sure when but before 2-3 days, if I remember right.

HAEdwin, 2024-02-15:
Weird,... in 1.4.0 there was no change that could have caused this. I will also look into it.

HAEdwin, 2024-02-16:
For me Today Energy in HA was 1,22kWh on the APSysems EMA website 1,21kWh. The integration can't be wrong, this data is just being read from the ECU which holds the data. There is no extra calculation of some sort. Was the ECU-B firmware updated recently? Since the latest 2024.2.2 release of HA, you are able to export data from the ECU Current Power graph and compare it with the exported data from the EMA website. Can you show this data?

RkcCorian, 2024-02-16:
@HAEdwin : No, it was really a problem on HA side as ioBroker had at the same point of time the correct values. Therefore technically it cannot be on ECU side and ioBroker was matching with it.
Now, after the restart of HA yesterday it seems to work fine till now. Might be also a problem which is caused by the integration update?

HAEdwin, 2024-02-18:
@RkcCorian  Could be. Most of the times I am unable to reproduce the issues. There are 450+ installations of this integration, expecting that not all of them work for some reason can be expected. I can't always put my finger on why this might be.

christianvaes, 2024-02-18:
I think i found the initial reason where this issue was opened for. It seems to be wifi inference, as turning of the near wifi ap(netgear orbi mesh) solved the signal drops. However when I connect it again it starts again. Earlier this wasn't the case so i tried a bit more and it seems to be because of the powerstrip the ecu and the wifi ap are connected to. Once removing the powerstrip and connecting it right to the wallsocket the connection is almost stable(one signal drop today and one yesterday since I made the change). Not really sure what the exact reason for this is and will try with some other powerstrip(as I don't have enough power socktes now) but it seems to create huge inference between the zigbee of the ecu and wifi somehow.

Atleast this issue is not related to the integration at all

RkcCorian, 2024-02-18:
@HAEdwin : I can understand. Currently it is working fine 🙂

HAEdwin, 2024-03-30:
I'll close this issue, great that it is solved!

closed 2024-03-30 by HAEdwin