ksheumaker/homeassistant-apsystems_ecur #231: Different monthly total

Rooverz, 2024-02-13:
Hi,

I have different monthly total between home assistant and the ecu app. How can this be explained? 
[image]
[image]

HAEdwin, 2024-02-13:
Can you put the integration in debug mode and show me the ECU query data together with the monthly value at that point from what the ECU app displays?

HAEdwin, 2024-02-13:
You mean the EMA app, yes there is a difference because the data pushed to EMA might slightly differ from the data pulled from the ECU by the integration. These two moments in time of push and pull causes different data to be displayed.

Rooverz, 2024-02-14:
Hi, this morning there still Is the same difference. I have  put the integration in debug mode, where can I find the information you asked for?

HAEdwin, 2024-02-14:
If you stop debugging, the browser will automatically initiate a download of the logfile. But as I allready explained, due to the update difference between pull from the integration and push from the ECU to the EMA site there allways will be a difference. If you want to be sure about returned energy, the best way to measure is at the electricity meter.

closed 2025-05-02 by HAEdwin