ksheumaker/homeassistant-apsystems_ecur #233: Bug Solar Production (history?)

brunnomcosta, 2024-02-16:
Hello everything is fine? I managed to integrate and can see the data in real time in Home Assistant. However, I'm trying to integrate with the HA solar energy session and it doesn't work. I noticed that the statistics are empty, although the entity reports the value in real time. Other device statistics are keeping history correctly. But those integrated with ApSystems are not keeping history. Can you help me?

brunnomcosta, 2024-02-16:
[Uploading 2024-02-16 13-55-02.zip…]()

HAEdwin, 2024-02-23:
In the Energy Dashboard it can take a while before data arrives.
[image]

closed 2024-03-14 by HAEdwin