ksheumaker/homeassistant-apsystems_ecur #234: Unable to get correct data from ECU, and no cached data. See log for details, and try power cycling the ECU.

FrcJules, 2024-02-23:
Hello i have this error and to correct it i have to restart my ECU do you have any other idea to correct him
thanks 

`Logger: custom_components.apsystems_ecur
Source: custom_components/apsystems_ecur/__init__.py:140
Integration: APSystems PV solar ECU (documentation, issues)
First occurred: 12:16:57 (3 occurrences)
Last logged: 12:17:53

Using cached data from last successful communication from ECU. Exception error: can only concatenate str (not "NoneType") to str`

FrcJules, 2024-02-29:
I don't understand the ECU site works anyway

HAEdwin, 2024-02-29:
But have you initially installed the integration succesfully? Did the integration discover the devices?

FrcJules, 2024-03-01:
no I just installed the integration then added the ECU IP.

this two errors generated are due to the addition of the ecu to the integration
`Cette erreur provient d'une intégration personnalisée

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/config_flow.py:42
Integration: APSystems PV solar ECU (documentation, issues)
First occurred: 18:42:59 (1 occurrences)
Last logged: 18:42:59

APSystemsInvalidData exception: timed out
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 91, in open_socket
    self.sock.connect((self.ipaddr, self.port))
TimeoutError: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 42, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 98, in query_ecu
    self.open_socket()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 94, in open_socket
    raise APSystemsInvalidData(err)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: timed out
`
and 
`Logger: aiohttp.server
Source: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:421
First occurred: 18:42:59 (1 occurrences)
Last logged: 18:42:59

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 91, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 227, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 235, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 149, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 72, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 185, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 358, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 404, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 508, in _async_handle_step
    if result.get("preview") is not None:
       ^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'
`

HAEdwin, 2024-03-01:
Make sure to test the connection before configuring the integration
https://github.com/ksheumaker/homeassistant-apsystems_ecur?tab=readme-ov-file#test-your-connection-and-find-your-ecu-on-the-lan
What type of ECU and connection to your network do you currently have?

FrcJules, 2024-03-02:
I don't understand, I can't check the connection of the computer with nc but I have to connect my extension to the ECU
[image]
[image]

HAEdwin, 2024-03-02:
Seems the integration is set up correctly, all should work fine. With nc you are also able to connect but reopened the port whilst it was allready opened. Have you put  the entities on the dashboard allready?

FrcJules, 2024-03-02:
I can't use "APS1100160001END" but the integration seems to work, I managed to add entities to my dashboard so I hope it will work

HAEdwin, 2024-03-02:
Great, I'll close this issue since it is not really an issue but a configuration issue. Feel free to open a new issue if needed.

closed 2024-03-02 by HAEdwin