ksheumaker/homeassistant-apsystems_ecur #235: Energy today wrong

RackerCRO, 2024-02-25:
I've noticed for some time now that Energy Today is higher in Home Assistant than reported on AP app or AP web site. 
For example, HA for today reports 9.5kWh production where AP app only 7.9kWh. I've checked for past 7 days and it's higher for every day. I tried rebooting my ECU but it didn't help. My ECU is R model 2160
[image]
[image]

Since I'm checking these values during the night and for past days this can't be delay issue I guess. 

Regards.

HAEdwin, 2024-02-25:
Yes, you are right. When I connect directly to my ECU (2160) the ECU app shows the same value (as expected) as the integration which is currently higher than the EMA website. I don't know what's going on or what data to trust right now; the value from the ECU itself or the EMA site. Btw. Today Energy is not being send to EMA, Today Energy is calculated locally on the server. It's certainly not an integration issue, the integration is just reading the Today Energy parameter from the ECU. I'll have to spend some time to perform a parallel calculation and see which value is correct. What firmware are you running?

RackerCRO, 2024-02-25:
My EMA app has same values as EMA website.
[image]

Where can I check FW version? EMA app manager?

HAEdwin, 2024-02-25:
It's an attribute of Today Energy
[image]

The EMA app takes data from the EMA site, the ECU app or ECU manager app from the ECU.

RackerCRO, 2024-02-25:
Fond it
[image]

HAEdwin, 2024-02-26:
Currently Today Energy matches in EMA and ECU, both 1,23kWh

RackerCRO, 2024-02-26:
Not here :(
EMA app
[image]

HA
[image]

HAEdwin, 2024-02-26:
Strange, can't do anything about this right now.

HAEdwin, 2024-03-14:
@RackerCRO I'll close this issue for I can't solve this. Impact is huge thought taking all the APsystems customers worldwide. Apparently it was never the intention to follow the ECU values ​​so closely, all customers look at the EMA site. Feel free to raise a new issue if any other arrises.

closed 2024-03-14 by HAEdwin