ksheumaker/homeassistant-apsystems_ecur #238: Connection fails since today's app update

eb-stefan, 2024-03-08:
I have a new app since today and can no longer connect to the ECU-B (port 8899 can no longer be opened)

HAEdwin, 2024-03-08:
What do you mean by "I have a new app"?

eb-stefan, 2024-03-08:
The “EMA app” looks a little different and asked me about new policies.

btw, In addition to HA, I also use Node-Red access directly to the TCP port. This no longer works either.

thomasvanhorde, 2024-03-08:
I have this error log 
2024-03-08 14:50:54.491 WARNING (SyncWorker_14) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: [Errno 113] Host is unreachable

But in https://www.apsystemsema.com/ all is OK

thomasvanhorde, 2024-03-08:
Hi,
I restart ECU and now is ok ! Just waiting tomorrow for datas but is connected to Hass

HAEdwin, 2024-03-10:
@eb-stefan Did you allready restart the ECU or do you still have isseus connecting?

HAEdwin, 2024-03-17:
@thomasvanhorde @eb-stefan  Everything still in working order?

thomasvanhorde, 2024-03-17:
> @thomasvanhorde @eb-stefan Everything still in working order?

Yes, thank you ! All is okay after reboot ECU 

[image]

eb-stefan, 2024-03-20:
Sorry, I was unfortunately sick last week.
Everything is working fine again.

What solution do you have? I have moderate experience with Node-Red :-)

HAEdwin, 2024-03-20:
Hope all is well again. I'm still working on the proxy method. Before the EU is sending the data to EMA it passes the proxy integration to register data in Home Assistant. Disadvantage is that signal data is not present, advantage is that current en voltages of each panel is now visible. (have to do some manual calculations that hopefully will get closer to the values EMA calculates. The ECU values are less accurate.
[image]
I'll close this issue, feel free to open a new one when ever needed.

closed 2024-03-20 by HAEdwin

eb-stefan, 2024-03-20:
I'm currently using a TCP function in NR. It has been running stable for 3 years.
I've only been working with HA for a few weeks. But my NR script is similar to this one:
https://community.home-assistant.io/t/apsystems-aps-ecu-r-local-inverters-data-pull/260835