ksheumaker/homeassistant-apsystems_ecur #239: Reset ECU by sending CMD

jmplonka, 2024-03-09:
It would be really cool to send a reset command to the ECU.
Just on March 8th my ECU-B did't show any values and was not reachable. Only resetting it be switching off and on again did the trick.
But if I'm not able to repower the ECU it would be helpful to send a reset command from Home Assistant.
Unfortunately I wasn't able to find the CMD-sequence.

Thanks

ps: awesome job till now 👍

HAEdwin, 2024-03-11:
For the ECU-B and older model ECU-R (2160xxxxx) there is not yet discovered a way to soft reset the ECU unfortunately, so a smart powerplug seems the only sollution. For the later ECU-R and ECU-C there is a soft reset method found that can be used with the integration.

HAEdwin, 2024-03-14:
@jmplonka I'll close this issue - might any other question arrise, feel free to open a new one.

closed 2024-03-14 by HAEdwin