ksheumaker/homeassistant-apsystems_ecur #240: ECU-3 Support

CayCanSafety, 2024-03-11:
Hello,

I can't see anything about this integration supporting ECU-3. (Specifically Version 4). It's connected via ethernet on the LAN. Anything you need from me to help support?

HAEdwin, 2024-03-12:
No extensive research into compatibility has ever been done (https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/156). It is therefore unclear whether the command set for the ECU-3 is identical to the other ECU models. If I remember correctly, the ECU-3 can also be configured to function via WiFi. If you want to try to see if you get a response from the ECU-3 with the connection test using both Ethernet and WiFi: https://github.com/ksheumaker/homeassistant-apsystems_ecur#test-your-connection-and-find -your-ecu-on-the-lan

https://github.com/jeeshofone/ha-apc-ecu-3 might work for you.

HAEdwin, 2024-03-14:
@CayCanSafety I'll close this issue. Feel free to open a new issue if you've found a way to use this integration with the ECU-3 or have any other question.

closed 2024-03-14 by HAEdwin