ksheumaker/homeassistant-apsystems_ecur #241: Not updating energy production figures, but correctly updating inverter data (curr. power, voltage etc).

IgorQuint, 2024-03-13:
Hi there,

I've installed the APS integration yesterday and somehow it only updated the lifetime energy production once at the first connection. 

Particularly, energy production today remains at zero and lifetime remains at 18,6kwh (static).
All other attributes like voltage, inverter temp, current power are updated nicely every 5minutes, as to be expected given the 300s query interval. What may cause this behaviour? 

I am using a ECU-R which I have connected wireless. Used the terminal and can properly connect to the ECU (which also has a fixed IP). Thank you!

HAEdwin, 2024-03-13:
Strange, only thing I can imagine is that you've somehow created double entities and was looking at inactive onces. Easiest way to check is going to: Settings > Devices & Services > Select the devices tab > select the APS devices one by one to display the Device Info Page and see if their entities are updating.

IgorQuint, 2024-03-13:
> Strange, only thing I can imagine is that you've somehow created double entities and was looking at inactive onces. Easiest way to check is going to: Settings > Devices & Services > Select the [...]

Thanks. I did a quick check. Nothing indicating that there are double entities. I have a ECU device alongside two (QT2) inverters. 

See attached some screens I took. Very strange indeed.. For some reason my debug logs aren't downloading so I cannot upload them at this time. 

[image]
[image]

HAEdwin, 2024-03-13:
I see you have the ECU_R_1.3.10 version, I'm at 1.2.28 with the same type of ECU (2160xxxx). APSystems never communicates the releasenotes going with the new ECU firmware releases so there might be going on something I can't put my finger on...
Alternative way to show the logs is to install: Settings > Add-ons > File Editor
[image]

Then browse to: home-assistant.log. I am particularly interested in the raw data that's produces when debugging is on.

IgorQuint, 2024-03-13:
2024-03-13 21:17:06.493 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303934303030313231363030303334313734353031000000ba0000000000000000d0d0d0d0d0d0d00002000031303031324543555f525f312e332e31303030394574632f474d542d3880971b09f4d5000000000000454e440a'
2024-03-13 21:17:16.731 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303038333030303230303031000220240313211236901500034029003032000000640000018200000183000101850000901500034411003032000000640000018700000186000101830000454e440a'
2024-03-13 21:17:16.732 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303033323030333030309015000340290090150003441100454e440a'
2024-03-13 21:17:16.733 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Got data from ECU
2024-03-13 21:17:16.733 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Returning {'timestamp': '2024-03-13 21:12:36', 'inverters': {'901500034029': {'uid': '901500034029', 'online': False, 'signal': 0, 'frequency': 0.0, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [0, 0, 1, 0], 'voltage': [386, 387, 389]}, '901500034411': {'uid': '901500034411', 'online': False, 'signal': 0, 'frequency': 0.0, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [0, 0, 1, 0], 'voltage': [391, 390, 387]}}, 'today_energy': 0.0, 'ecu_id': '216000341745', 'lifetime_energy': 18.6, 'current_power': 0, 'qty_of_inverters': 2, 'qty_of_online_inverters': 0, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}
2024-03-13 21:17:16.737 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.350 seconds (success: True)
2024-03-13 21:17:16.738 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - current_power
2024-03-13 21:17:16.739 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for current_power
2024-03-13 21:17:16.740 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - today_energy
2024-03-13 21:17:16.740 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for today_energy
2024-03-13 21:17:16.741 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - lifetime_energy
2024-03-13 21:17:16.741 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for lifetime_energy
2024-03-13 21:17:16.745 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_inverters
2024-03-13 21:17:16.745 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_inverters
2024-03-13 21:17:16.746 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_online_inverters
2024-03-13 21:17:16.746 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_online_inverters
2024-03-13 21:17:16.747 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2024-03-13 21:17:16.747 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2024-03-13 21:17:16.747 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2024-03-13 21:17:16.747 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2024-03-13 21:17:16.748 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-03-13 21:17:16.748 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-03-13 21:17:16.749 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-03-13 21:17:16.749 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-03-13 21:17:16.753 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-13 21:17:16.754 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-13 21:17:16.754 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034029 0
2024-03-13 21:17:16.754 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-13 21:17:16.755 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-13 21:17:16.755 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034029 1
2024-03-13 21:17:16.755 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-13 21:17:16.755 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-13 21:17:16.756 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034029 2
2024-03-13 21:17:16.756 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-13 21:17:16.756 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-13 21:17:16.756 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034029 3
2024-03-13 21:17:16.757 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2024-03-13 21:17:16.757 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
[... cut ...]
2024-03-13 21:22:31.768 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-13 21:22:31.768 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034411 2
2024-03-13 21:22:31.769 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-13 21:22:31.769 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-13 21:22:31.769 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034411 3

IgorQuint, 2024-03-13:
This is some of the raw output. seems to repeat every 5 min on the query

HAEdwin, 2024-03-13:
Not much to see right now. Fact is that this integration only reads what is available from the ECU so if de ECU is wrong this data is wrong. Keep an eye on it at daytime. Besides this integration polling each five minutes, the ECU also pushes data to the EMA site every five seconds. Data from the ECU is only renewed every 5 minutes.

IgorQuint, 2024-03-14:
Not entirely sure what changed, but I did just get an updated to these two entities (energy lifetime & today). First update at 10.55, at 11.00 also succesful. Note that the current power entities (alongside others) did got updated earlier today once the sun got up. Very strange, but happy nontheless! :)

2024-03-14 10:50:51.387 DEBUG (SyncWorker_46) [custom_components.apsystems_ecur] Querying ECU...
2024-03-14 10:50:56.417 DEBUG (SyncWorker_46) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303934303030313231363030303334313734353031000000ba0000030f00000003d0d0d0d0d0d0d00002000231303031324543555f525f312e332e31303030394574632f474d542d3880971b09f4d5000000000000454e440a'
2024-03-14 10:51:06.662 DEBUG (SyncWorker_46) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303833303030323030303100022024031410473690150003402901303201f3007700470197004701990049019e004990150003441101303201f3007c0078019b007a019a007f019d007e454e440a'
2024-03-14 10:51:06.663 DEBUG (SyncWorker_46) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303332303033303030901500034029ab901500034411a9454e440a'
2024-03-14 10:51:06.664 DEBUG (SyncWorker_46) [custom_components.apsystems_ecur] Got data from ECU
2024-03-14 10:51:06.664 DEBUG (SyncWorker_46) [custom_components.apsystems_ecur] Returning {'timestamp': '2024-03-14 10:47:36', 'inverters': {'901500034029': {'uid': '901500034029', 'online': True, 'signal': 67, 'temperature': 19, 'frequency': 49.9, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [71, 71, 73, 73], 'voltage': [407, 409, 414]}, '901500034411': {'uid': '901500034411', 'online': True, 'signal': 66, 'temperature': 24, 'frequency': 49.9, 'model': 'YC1000/QT2', 'channel_qty': 4, 'power': [120, 122, 127, 126], 'voltage': [411, 410, 413]}}, 'today_energy': 0.03, 'ecu_id': '216000341745', 'lifetime_energy': 18.6, 'current_power': 783, 'qty_of_inverters': 2, 'qty_of_online_inverters': 2, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}
2024-03-14 10:51:06.668 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.281 seconds (success: True)
2024-03-14 10:51:06.669 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - current_power
2024-03-14 10:51:06.669 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for current_power
2024-03-14 10:51:06.671 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - today_energy
2024-03-14 10:51:06.671 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for today_energy
2024-03-14 10:51:06.672 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - lifetime_energy
2024-03-14 10:51:06.672 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for lifetime_energy
2024-03-14 10:51:06.677 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_inverters
2024-03-14 10:51:06.677 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_inverters
2024-03-14 10:51:06.678 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_online_inverters
2024-03-14 10:51:06.678 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_online_inverters
2024-03-14 10:51:06.678 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2024-03-14 10:51:06.679 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2024-03-14 10:51:06.679 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2024-03-14 10:51:06.679 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2024-03-14 10:51:06.680 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-03-14 10:51:06.680 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-03-14 10:51:06.680 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-03-14 10:51:06.680 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-03-14 10:51:06.681 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-14 10:51:06.681 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-14 10:51:06.681 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034029 0
2024-03-14 10:51:06.682 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-14 10:51:06.682 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-14 10:51:06.683 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034029 1
2024-03-14 10:51:06.683 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-14 10:51:06.683 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-14 10:51:06.683 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034029 2
2024-03-14 10:51:06.684 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
[... cut ...]
2024-03-14 11:06:51.528 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-14 11:06:51.528 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034411 2
2024-03-14 11:06:51.529 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-03-14 11:06:51.529 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-03-14 11:06:51.529 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  901500034411 3

HAEdwin, 2024-03-14:
I guess it was because at the end of the day Lifetime Energy won't increase much. Integration should work (it does for 400+ users). Allright to close this issue? The purpose of the switch is turning on/off the query it will not have any effect on trying to solve anything. Originally the switch was used to turn off queries at night.

IgorQuint, 2024-03-14:
> I guess it was because at the end of the day Lifetime Energy won't increase much. Integration should work (it does for 400+ users). Allright to close this issue? The purpose of the switch is turning [...]

hmm yes, tried it with hope that it would force the query, but didn't expect much. 

I was having the issue throughout the day and also the day before (There was actually some energy production yet the entity didn't update), so it was not me merely looking at nighttime. However, it works now and happy about that! So whatever it was, issue can be closed. Thanks for helping out! 👍

HAEdwin, 2024-03-14:
Indeed hard to tell what was going on. State was called every time and dictionary showed the presents of the parameter. Weird ECU glitch or something... No problem glad you like the integration. If you see improvements in the code or like to help with building a proxy integration to obtaining the data, let me know. I'll close the issue - feel free to open a new one when needed.

closed 2024-03-14 by HAEdwin