ksheumaker/homeassistant-apsystems_ecur #244: ECU-R (2160xxxxxxx) connection refused and requiring manual reboot often

gwilliams1205, 2024-03-19:
I just started using this integration and am very grateful for it.. Though for whatever reason it doesn't seem to be very stable in my environment.. I am running into cached mode A LOT: 
[image]

and have set an automation to cycle a smart plug to reboot the device.. it has automatically done it 3 times today.. with the last time taking nearly 2 hours to reconnect.. My wireless signal is great.. I have tried some of the tricks by moving the system and even removing the wifi antenna, but it doesn't seem to be helping..

This is happening in the middle of the day not after sunset..

Is there anyway to make this system work with a wired connection?

HAEdwin, 2024-03-20:
That's a lot indeed! I own the same ECU with firmware. It is is hard to tell why this is happening could be interference from surrounding WiFi and/or Zigbee signals. Try to switch channels on the 2.4Ghz band or see if the ECU is roaming between AP's. It is true that sometimes it happens more frequent than other periods but this is extreem. Wired connection does not work, feature is undocumented. Here's my cache usage:
[image]

gwilliams1205, 2024-03-20:
I moved my device again.. it's right under the WAP and has fantastic signal.. but it didn't seem to help.. It is going into cached mode crazy often.. I have also changed my 2.4ghz channel.. I don't know if it's just my unit or what's going on.. But  have no idea why it would keep getting a connection refused

HAEdwin, 2024-03-20:
Preventing interference sometimes means weaken the signal of surrounding devices (Zigbee and WiFi share the same channel), not to strengthen it. So removing the ECU further from the AP and other devices that might cause interference could be considered. Take in account the signal strength of the inverter (monitor that).

gwilliams1205, 2024-03-20:
Is there anyway to test to see what's going on root cause wise? 

Signal wise things look good, and I I can see I'm not losing packets via network.. I'd love to know if the system itself is overworked or something.. 

Any issues with me switching the polling to a shorter time to see if it is still switching over to cache mode as much? Or will that make matters worse?

HAEdwin, 2024-03-20:
Keep an eye on the signal data from the inverters. Increasing the number of queries will only make things worse, an interval of 300 seconds should be fine but you could try 240. Try a restart of the ECU if you've increased the query interval before.

gwilliams1205, 2024-03-20:
Update… I’ve rebooted both the ECU and my HA container.. since back up I haven’t had an issue. I saw some issues with another device automation via HA and so I reset it. It’s been over 2 hours with no caching. It’s a good start. If we make it through the next day or two with little to no cached mode I think we are good

HAEdwin, 2024-03-30:
I'll close this issue for now. Feel free to raise another issue when needed.

closed 2024-03-30 by HAEdwin