ksheumaker/homeassistant-apsystems_ecur #251: Configuration

alain1966, 2024-05-02:
Good morning. I can't configure the integration. I enter the IP address of my Ecu C: 172.30.1.1, what is the password that I must enter, that of my SSID (Box) or the password of the EMA application on the net? I tried both but it doesn't work. Thanks for your help

gbetous, 2024-05-03:
Try without any password, I think that was the case for me (not sure but always worth a try !)

HAEdwin, 2024-05-03:
Hi Alain, I added some additional info to the configure instructions: _The optional fields [Specify SSID] and [Specify password] are the parameters you would like to assign to the ECU when an automatic software restart is being applied (for ECU-R (sunspec) and ECU-C models only). These parameters are not being used during setup. If setup fails, in most cases the wrong IP-address was specified._ 

The software reset is based on changing the configuration which demands a software reset from the ECU. You can specify the initial parameters for your situation.

For your ECU-C make sure you specify the IP-Address assigned to the wired ethernet connection.

alain1966, 2024-05-03:
hello thank you for your feedback. I will connect the ECU wired to see....🤞🏼

HAEdwin, 2024-05-03:
If you own an ECU-C you should indeed use the wired connection see: https://github.com/ksheumaker/homeassistant-apsystems_ecur?tab=readme-ov-file#prerequisites
The Wifi connected ECU does not expose port 8899

jsalva23, 2024-05-23:
> If you own an ECU-C you should indeed use the wired connection see: https://github.com/ksheumaker/homeassistant-apsystems_ecur?tab=readme-ov-file#prerequisites The Wifi connected ECU does not expose port 8899 required for this integration to work.

Hey HAEdwin,

is there a way to have this work on a ECU-R but with a wire connection? I have it configure like that because the location where I have the ECU does not have wireless access, and so its impossible for me to access it via wireless, its just a different port for wire? or is is more complex than that?

Thanks !

HAEdwin, 2024-06-05:
No there's not unless a new firmware update will support that.

closed 2024-06-05 by HAEdwin