ksheumaker/homeassistant-apsystems_ecur #253: Using cached data a lot

m3tc0n, 2024-05-16:
Hi, my APsystems ECU integration is using cached data a lot. I don't think that's any good. How to solve that? It's occurring on version 1.4.1 and beta version 1.4.2.

[image]

HAEdwin, 2024-05-17:
Please read earlier messages about this: https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues?q=is%3Aissue+is%3Aclosed+ecu+using+cached+data 
Is more an ECU placement issue than it is an integration issue.

m3tc0n, 2024-05-17:
Thanks for your response. My ECU has been in the same place as long as I have it. It used to work fine, but lately I have this problem.

HAEdwin, 2024-05-17:
Look for environmental changes you made since the cache errors occured. Have you added an extra AP, Zigbee devices? Does it help when you position the ECU somewhere else?

m3tc0n, 2024-05-17:
I didn't make any environmental changes. I now have replaced my ECU to the attic, but that makes no difference.

HAEdwin, 2024-05-23:
Great! Thanks for the feedback, I'll close the issue. Feel free to open a new one if needed.

closed 2024-05-23 by HAEdwin