ksheumaker/homeassistant-apsystems_ecur #257: Unable to connect

chuppito, 2024-05-25:
Hello, I haven't been able to connect for some time. I deleted and reinstalled the integration but it no longer works 
[image]
[image]

nils985, 2024-05-27:
Hello. got the same problem after updating the HA-Core to 2024.5.5. Restoring  backup (HA-Core 2024.5.4) didn´t helps.

Rooverz, 2024-05-28:
Hello,

I also have the same problem. 

[image]
[image]

KPI33, 2024-05-28:
Same problème here
HA 2024.05.24 and ecu b

Rooverz, 2024-05-28:
Hi,

I have managed to solve the problem by changing the Ip adres in the configuration of the aps integration in HA. Apperantly the ip of my ECU was changed. Hope this helps!

chuppito, 2024-05-28:
I have the right IP address, impossible for me

KPI33, 2024-05-28:
> Hello, I have the right IP address, impossible for me

Same i am sure with thé right ip

nils985, 2024-05-29:
Update:
I have managed to establish a connection to my two ECU-Bs.
- After restoring the HA-Core 2024.5.4, I uninstalled the ECU-R service. 
- System reboot
- Then updated to HA-Core 2024.5.5 
- System reboot
- Installed the ECU-R service again via HACS
- System reboot
- ECU-R service set up (ip etc.)
- System reboot

chuppito, 2024-05-29:
Don't work for me

chuppito, 2024-05-30:
1.4.3 don't work

HAEdwin, 2024-05-31:
From

closed 2024-05-31 by HAEdwin

chuppito, 2024-05-31:
There was the problem. I had a fixed IP address from the start, I don't know why it no longer worked

n4rs, 2024-06-19:
> Hi, I have managed to solve the problem by changing the Ip adres in the configuration of the aps integration in HA. Apperantly the ip of my ECU was changed. Hope this helps!
> […](#)

What file did you change and where?