ksheumaker/homeassistant-apsystems_ecur #260: Local access only

bruvv, 2024-07-29:
I tried to disable the internet access to my ecu_b but home assistant does not receive anymore data. Is this addon depending on the AP System cloud for it's data instead of local pulling?

HAEdwin, 2024-07-29:
Limiting internet access to your ECU causes the ECU to crash and as a result won't respond to queries from the integration. The integration is not depending on the AP Systems cloud.

bruvv, 2024-07-29:
that is so sad to hear... any idea on how we can make it just loop back to itself thinking he is online?

HAEdwin, 2024-07-29:
Yes, I've worked on that but I have never tested this extensively over a long period of time because I would miss my data from the EMA server :) It is not that complicated to send the correct answers to the ECU, but it involves a little more than a loopback or an "ok".

bruvv, 2024-07-29:
so nothing simple we can do in unifi with this ;) ?

tannercollin, 2024-08-15:
I've been able to run it isolated / offline for a few years now without problems. This is the response that it expects:

https://git.tannercollin.com/tanner/solar-display/src/branch/master/server/main.py#L25

You can see that TCP server listens on ports 8995, 8996, and 8997. You'll need to override your local DNS to point the domains to your own server. For example, with dnsmasq:

```
address=/.apsystemsema.com/192.168.0.106
address=/.apsema.com/192.168.0.106
```

I also have to use a timer outlet to power cycle the ECU nightly.

HAEdwin, 2024-08-17:
Look at https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy as an (still EMA dependent) alternative but it is unfinished work, lack of time - but since I've suggested this for a view years back I thought it would be time to publish this, I recognise the code :) Basics for this integration were build by me but Mark Parker was so kind to speed things up in development with his huge knowledge of Python and HA. See how this works as an alternative and I might spend some time on it.
@tannercollin In the past, the 101 response was not sufficient on the long run but firmware changes might have overcome this.

HAEdwin, 2025-01-29:
@bruvv I might extend the integration at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy with an option which makes the ECU think it is online. If you still like to have this option, feel free to open the question there. I'll close this issue here for I'm soon no longer maintaining this version and continued at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader

closed 2025-01-29 by HAEdwin