ksheumaker/homeassistant-apsystems_ecur #261: Integration works for one day.

Dakhaasje, 2024-08-04:
I have :
 Core 2024.7.4
Supervisor 2024.06.2
Operating System 12.4
Frontend 20240710.0
1 ECU-B, with 2 DS3L inverters
Every day this works IF I reboot HA... rebooting the ECU-B has no effect at all.

HAEdwin, 2024-08-04:
Have you allready investigated the logs?

Dakhaasje, 2024-08-06:
there is nothing in the log with ' ECU' the next day.

Dakhaasje, 2024-08-06:
`024-08-06 05:09:43.413 DEBUG (SyncWorker_16) [custom_components.apsystems_ecur] Not querying ECU due to query=False
2024-08-06 05:09:43.415 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 0.002 seconds (success: True)
2024-08-06 05:09:43.415 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - current_power
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for current_power
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - today_energy
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for today_energy
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - lifetime_energy
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for lifetime_energy
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_inverters
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_inverters
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_online_inverters
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_online_inverters
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000980023 0
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.421 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.421 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000980023 1
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000812560 0
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000812560 1
2024-08-06 05:14:43.414 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Not querying ECU due to query=False
2024-08-06 05:14:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 0.003 seconds (success: True)
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - current_power
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for current_power
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - today_energy
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for today_energy
[... cut ...]
2024-08-06 06:59:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-08-06 06:59:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-08-06 06:59:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-08-06 06:59:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-08-06 06:59:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power`

Dakhaasje, 2024-08-06:
```
> 024-08-06 05:09:43.413 DEBUG (SyncWorker_16) [custom_components.apsystems_ecur] Not querying ECU due to query=False
2024-08-06 05:09:43.415 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 0.002 seconds (success: True)
2024-08-06 05:09:43.415 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - current_power
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for current_power
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - today_energy
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for today_energy
2024-08-06 05:09:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - lifetime_energy
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for lifetime_energy
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_inverters
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_inverters
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_online_inverters
2024-08-06 05:09:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_online_inverters
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2024-08-06 05:09:43.418 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-08-06 05:09:43.419 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000980023 0
2024-08-06 05:09:43.420 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.421 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.421 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000980023 1
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2024-08-06 05:09:43.422 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000812560 0
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2024-08-06 05:09:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  703000812560 1
2024-08-06 05:14:43.414 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Not querying ECU due to query=False
2024-08-06 05:14:43.416 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 0.003 seconds (success: True)
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - current_power
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for current_power
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - today_energy
2024-08-06 05:14:43.417 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for today_energy
[... cut ...]
2024-08-06 06:59:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2024-08-06 06:59:43.423 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2024-08-06 06:59:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2024-08-06 06:59:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2024-08-06 06:59:43.424 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
```

Dakhaasje, 2024-08-06:
2024-08-06 06:09:43.413 DEBUG (SyncWorker_31) [custom_components.apsystems_ecur] Not querying ECU due to query=False

Dakhaasje, 2024-08-06:
> If query fails for a predefined number of times the switch will switch off so that an automation can react on it (optionally reboot, send a notification, ..)

all night it fails... I added an automation at 1 hour after sun-up, to set it to ON if not already ON

Dakhaasje, 2024-08-06:
> If you encounter this often, try to reposition the ECU away from possible interference from other devices especially WiFi AP's or other Zigbee devices (you allready mentioned that in the other thread).
The Ecu-B sits lonely on a roof beam in the shed with an unobstructed view on the panels with outdoor extended antenna.

HAEdwin, 2024-08-07:
How often does the integration use cached data (ECU Using Cached Data entity)? Mine looks like: 
[image]
which is normal behaviour...

Dakhaasje, 2024-08-08:
> How often does the integration use cached data

[image]
, but the turning off/on automation made it work for 2 days in a row
[image]

HAEdwin, 2024-08-12:
Some have positive experiences by removing the WiFi antenna from the ECU, I positioned the antennas like so \  / (facing away from each other). There is no exact solution, you'll have to try out some to get the optimal results. If you agree I'll close this issue for it is not an integration issue.

Dakhaasje, 2024-08-12:
> Some have positive experiences by removing the WiFi antenna from the ECU,
I pointed one in (90 degrees to) the direction of the wifi AP, the other I put in  clear view of the solar panels, using about 30cm extension cable leading outside. 78% signal should be fine.

Dakhaasje, 2024-08-12:
> , but the turning off/on automation made it work for 2 days in a row

I actually made the ECU query turn on after the Enphase started giving power... Nothing beats Enphase ;-)

HAEdwin, 2024-08-21:
You might also want to try https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy 
As is often the case, a relatively small number of users will have problems with the integration, but with over 600 active installations, I think the integration is doing something right.

Dakhaasje, 2024-08-21:
> I think the integration is doing something right.

I agree

Dakhaasje, 2024-08-21:
[image] 1 relapse, possibly moment of HA restart

Dakhaasje, 2024-08-21:
backup automation works for me

alias: Als ECU query device uitgezet, terwijl er nog zon is, zet weer aan
description: ""
trigger:
  - platform: device
    type: turned_off
    device_id: 1b1e #(query device)
    entity_id: 21cd88fd5cb2c
    domain: switch
    for:
      hours: 0
      minutes: 1
      seconds: 0
condition:
  - condition: state
    entity_id: input_boolean.enphase_werkt #(when other system still works, this helper is ON)
    state: "on"
  - condition: sun
    before: sunset
    after: sunrise
action:
  - action: script.ecu_query_aanzetten #(switch query back on)
    metadata: {}
    data: {}
mode: single
``

HAEdwin, 2024-08-21:
Great solution. I let mine (an older ECU-C starting with 2160...) query 24/7 and set the integration to try 5 times after that, the ECU Query Switch is switched off. So now my automation can react on that with a reboot using a smart powerplug AND I turn the ECU Query Switch back on (like you did also). 

```
2024-08-19 13:16:57.841 WARNING (SyncWorker_55) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: [Errno 111] Connection refused
2024-08-19 13:21:57.788 WARNING (SyncWorker_33) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: [Errno 111] Connection refused
2024-08-19 13:26:57.807 WARNING (SyncWorker_43) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: [Errno 111] Connection refused
2024-08-19 13:31:57.849 WARNING (SyncWorker_30) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Invalid data error: [Errno 111] Connection refused
2024-08-19 13:31:57.850 WARNING (SyncWorker_30) [custom_components.apsystems_ecur] Communication with the ECU failed after 4 repeated attempts.
2024-08-19 13:31:57.850 WARNING (SyncWorker_30) [custom_components.apsystems_ecur] Try manually power cycling the ECU. Querying is stopped automatically, turn switch back on after restart of ECU.
```
But this is needed once a month at max, more often the ECU fails to let it being queried one time and next time all is well again. 
[image]
This could be due to the fact that I have the proxy running simultanious, in fact there might be conflicts when the ECU also tries to upload data to EMA while at the same time the integration is querying the ECU but I never have matched the timestamps.

HAEdwin, 2025-01-20:
I'll close this issue for now. Placement of the ECU is key for keeping EMA and this integration up-to-date. If any other question arises, feel free to open another issue.

closed 2025-01-20 by HAEdwin