ksheumaker/homeassistant-apsystems_ecur #263: ecu_today_energy different from EMA

iomartin, 2024-08-29:
I installed this integration yesterday but noticed that the `sensor.ecu_today_energy` is quite different from what the EMA site reports. But it's not the double as reported in #245.

I have an ECU-R (2162xxxxxxxx) with DS-3 inverters
core 2024.8.3
supervisor 2024.08.0
operating system 13.1
Integration v1.4.3

For example, right now the integration says 8.38 kWH but the EMA is at 6.01 kWh.

I'm fairly new to HA, so please let me know what other info is required to debug this

HAEdwin, 2024-08-30:
There is indeed a difference in values but it is not due to the integration. I you check yesterdays Energy again you might notice that EMA has a different value because there are data corrections taking place after sundown. Also the calculation within the firmware apparently differs from what EMA calculates.

nioman86, 2024-09-11:
I have the same issue on a ECU-R (2160xxxxxxxx) The daily statistics are way off. 
Here is how the statistic looked for ecu_today_energy for 9th of this month:
[image]

This is how the same statistic shows in EMA:
[image]

For some reason the Home assistant seems to start collecting data everyday between 10 and 11 am.

This also effects the total calculation for the day. The EMA app shows a total solar power of 28,8 kWh and HA shows 24,8 kWh. The difference seems to be inconsistent as the largest difference was almost 10 kWh and the next day it was only 4 kWh:
[image]

HAEdwin, 2024-09-12:
You might want to check the time zone and locale settings in HA and the ECU. Also look at automations if you didn't accidently switch quering off. Compare the first values with the online status from the inverters, do they match up?

nioman86, 2024-09-12:
Checked the time zones and locale and they all match (Europe/Helsinki).
No automations in place. 
The time stamps are weird
At 6.38 The ECU inverter has turned online at sunrise (state 4) 
At 9.52 it gives a message of using cached data. At 10.13 cached data was disabled
At 10.35 first visible data on the ecu_today_energy

iomartin, 2024-09-13:
> There is indeed a difference in values but it is not due to the integration. I you check yesterdays Energy again you might notice that EMA has a different value because there are data corrections [...]

Indeed, after letting it run for a few days, the integration always match the EMA for previous days. It's unfortunate that it's off for for "live" values

mi-hol, 2024-09-14:
> It's unfortunate that it's off for for "live" values
> For example, right now the integration says 8.38 kWh but the EMA is at 6.01 kWh.

Is EMA maybe kind of "catching up" with HA during a few hours of delay?
Maybe you could collect the reported data from both sources in a 1 hour interval and share it here.
I could imagine that the current (apparently single global) EMA instance is kind of overloaded with the growing number of ECUs reporting to it.

HAEdwin, 2024-09-14:
Yes, that's what I explained earlier: There is indeed a difference in values but it is not due to the integration. I you check yesterdays Energy again you might notice that EMA has a different value because there are data corrections taking place after sundown. Also the calculation within the firmware apparently differs from what EMA calculates.

worker72, 2024-09-15:
What do you guys see when you compare the Total Energy values? 
I am now using my PV system for 2 months, and I see the Total Energy value in HA is higher than the value I see in the EMA app. 
The difference cannot be explained by calculations/compensations that take place at night, because that would create an offset. In my case, I see the differnce bewteen the two is slowly increasing (HA value is 6.6% higher than the value in the EMA app)

HAEdwin, 2024-09-15:
Do you mean Lifetime Energy or Today Energy? But it really doesn't matter - they both are different from EMA partly because the ECU query is not in sync with the push from the ECU to EMA, they sometimes can even clash if both happen at the same time. But anyway, the differences do facinate me and made me wonder if the proxy method would improve things but they don't (yet). I Guess that besides just checking if all inverters are online the best measure point is at the energy meter.

HAEdwin, 2024-10-28:
The difference cannot be resolved at this time. The proxy method also shows a difference while using the same data. This is mainly because the data has a time stamp attached to it. The data always covers the last 5 minutes. To my knowledge, Home Assistant is not able to store data with a time in the past. If an issue arises again, feel free to open a new issue.

closed 2024-10-28 by HAEdwin