ksheumaker/homeassistant-apsystems_ecur #264: ECU-R replaced, transfer to new entuty/sensor

foxtom22, 2024-08-30:
Hi,

Thank you for this add-on, very useful.

I am not very confortable with HA so excuse me if this request is not relevant here.
As the title says, my ECU was replaced because not connecting anymore to network (either RJ45 or wifi).
I updated the configuration in HA APSystem integration so good so far. Now the problem is I get some duplicated entities, like sensor.ecu_current_power and ensor.ecu_current_power2. 
Is there a way to transfer the data from the first entity to the new one so we still have the production history in the new one and therefore in the charts?

Thank you for reading !
Kind regards

HAEdwin, 2024-08-30:
Values of the new ECU should get updated in the new entities. Have you allready removed the old ECU? Settings > Integrations > "APSystems PV Solar ECU integration" > Devices > Select old ECU > then the three dots on top of eachother > Delete
This should also remove the associated entities.

HAEdwin, 2025-01-29:
@foxtom22 No I'm afraid you won't. The EMA site is the only one which keeps record of the complete history. The proxy method as wel as the reader method are just reading the data from the ECU, there is no platform like EMA to manage replacements. It would mean implementing a field in which you fill out the value for ECU Lifetime Energy and add that to the retreived data from the new ECU. If you still feel the need for it, post a message at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader because I'll stop maintaining this version. I'll close this issue for now.

closed 2025-01-29 by HAEdwin