ksheumaker/homeassistant-apsystems_ecur #265: ECU-R Not reporting Total and Daily Energy

mmornati, 2024-09-06:
**Integration Version**: 1.4.3
**APSystems Model**:ECU-R (2160xxxx)

Hey,
I've just installed the APSystems at home and thanks to this integration I could get "all" the information in home assistant. Appreciated, thanks.

I'm saying "all" because for a reasons I'm not able to fix, the produced energy reported is always 0.
[image]

A things I noticed is even when I connect to APManager App (locally) these information are 0, so I guess we are reading everything from there. But on the net and in the "end user application" everything is working correctly.

Does anyone have any tips on how to debug and fix it?

HAEdwin, 2024-09-08:
Having an ECU-R 2160xxxx with firmware 1.3.14 is odd - for this model I'm currently at ECU_R_1.2.31. Have you tried factory defaults (not even sure if this will downgrade the firmware). Ask APsystems for support because this is a firmware issue.

jan-dewit, 2025-01-07:
Hi @HAEdwin since I am experiencing this myself I came to this reported issue.
Actually I am running Firmware: ECU_R_1.2.31 and experiencing (partly) the same issue, the ECU Today Energy is not reporting any generated energy, while the ECU Lifetime Energy is increasing. The EMA App is actually reporting the Daily Energy generated, so this is kinda strange. I already power cycled the ECU but no luck, just stays 0,0 kWh.

jan-dewit, 2025-01-10:
@HAEdwin any tips/tricks I could try solving this?
Would love to get this working again as I am using this data for  some calculations and automations.

HAEdwin, 2025-01-10:
EMA does it's own calculations based on received data, that's the reason all seems fine there. 
This integration indeed reads data from the ECU the same way the EMA manager app or ECUapp does. So there is nothing I can do about that. Even when you would DM me the debug log, only conclusion would be that for some strange reason these values stay empty or zero. Btw APsystems might suggest that you only consult the EMA website or use their API. The method used in this integration was originally for installation and configuration only, not for continuous monitoring. As an alternative for 2160 models you could try https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy

jan-dewit, 2025-01-10:
Thanks for pointing me to the ecu_proxy repository, did not get earlier to my attention, configured it directly!

The debug log shows the following (altered the ecu and inverter ids):

```
2025-01-10 14:52:00.195 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Querying ECU...
2025-01-10 14:52:05.286 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303039343030303132313630303030363035393830310000d52d0000001f00000000d0d0d0d0d0d0d00002000231303031324543555f525f312e322e33313030394574632f474d542d3880971b01f5e9000000000000454e440a'
2025-01-10 14:52:15.424 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303037333030303230303031000220250110144922inverter_id101303101f3006b000800e7000900e7inverter_id201303301f3006a000200e7000200050005454e440a'
2025-01-10 14:52:15.424 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303332303033303030inverter_id1b5inverter_id2c6454e440a'
2025-01-10 14:52:15.425 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Got data from ECU
2025-01-10 14:52:15.425 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Returning {'timestamp': '2025-01-10 14:49:22', 'inverters': {'inverter_id1': {'uid': 'inverter_id1', 'online': True, 'signal': 70, 'temperature': 7, 'frequency': 49.9, 'model': 'YC600/DS3 series', 'channel_qty': 2, 'power': [8, 9], 'voltage': [231, 231]}, 'inverter_id2': {'uid': 'inverter_id2', 'online': True, 'signal': 77, 'temperature': 6, 'frequency': 49.9, 'model': 'QS1', 'channel_qty': 4, 'power': [2, 2, 5, 5], 'voltage': [231]}}, 'ecu_id': 'ecu_id', 'lifetime_energy': 5457.3, 'current_power': 31, 'qty_of_inverters': 2, 'today_energy': 0.0, 'qty_of_online_inverters': 2, 'data_from_cache': False, 'querying': True, 'restart_ecu': False}
2025-01-10 14:52:15.425 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.230 seconds (success: True)
2025-01-10 14:52:15.426 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - current_power
2025-01-10 14:52:15.426 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for current_power
2025-01-10 14:52:15.426 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - today_energy
2025-01-10 14:52:15.426 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for today_energy
2025-01-10 14:52:15.426 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class total_increasing - lifetime_energy
2025-01-10 14:52:15.426 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for lifetime_energy
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_inverters
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_inverters
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class None - qty_of_online_inverters
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for qty_of_online_inverters
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2025-01-10 14:52:15.427 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - signal
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for signal
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  inverter_id1 0
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2025-01-10 14:52:15.428 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  inverter_id1 1
2025-01-10 14:52:15.429 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - temperature
2025-01-10 14:52:15.429 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for temperature
2025-01-10 14:52:15.429 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - frequency
2025-01-10 14:52:15.429 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for frequency
2025-01-10 14:52:15.429 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - voltage
2025-01-10 14:52:15.429 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for voltage
[... cut ...]
2025-01-10 14:52:15.430 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2025-01-10 14:52:15.430 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  inverter_id2 2
2025-01-10 14:52:15.430 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State class measurement - power
2025-01-10 14:52:15.430 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] State called for power
2025-01-10 14:52:15.430 DEBUG (MainThread) [custom_components.apsystems_ecur.sensor] POWER  inverter_id2 3
```

HAEdwin, 2025-01-10:
I have checked the data and the frame composition still matches expectations. I expect that the EMA manager and ECUapp also show zero. I can't explain why, maybe corrupt firmware?

jan-dewit, 2025-01-14:
Hi @HAEdwin thanks for checking, the EMA App shows the daily generation, and the ECU Proxy integration is also showing the daily generation, I will use that for the time being, and keep monitoring to see if it will work again in the future.
Thanks for your work on the APsystems ECU Proxy integration, much appreciated!

HAEdwin, 2025-01-14:
Yes, both EMA and the Proxy integration do their own calculation. Unfortunately they don't match but use the same data. I'll close this issue. Feel free to open another if needed.

closed 2025-01-14 by HAEdwin