ksheumaker/homeassistant-apsystems_ecur #267: Improvement suggestion sent to APS

RenaudPerr, 2024-09-13:
Hi all
FYI, I have sent this mail to APSystem, hopping they will improve the local web site on ECU-C to read more frequently and easily last Produced/Consumed Power from CT Clamp.
Hopping this does not interfere with ksheumaker action plans, otherwise sorry for that.
BR

> Hello
> Please forward this mail to the R&D department or Product Improvement department of DS3/ECU-C products.
> 
> I am an IT engineer and I have 18 DS3 inverters and an ECU-C with two CT clamps, all supervised by Home Assistant.
> I can understand that for the DS3 production readings you limit the refresh period to 5 minutes (many inverters connected in Zigbee which is not very quick), but for the clamps I do not see why you do not lower this period to one minute as it is a simple Analog to Digital conversion. This would would allow me/Home Assistant to adjust more quickly my consumption in case of production drop or increase.
> In addition, it would be good to have in the local APSystem web application in the Advanced/Power page, directly the 9 values :Produced A/B/C, Consumed A/B/C and Imported A/B/C ​​provided by the clamps in addition to the curve.
> This would allow Home Assistant to simply read these values ​​(by web scraping for example) rather than having to look for the last points in the curves.
> I hope I have been clear in my requests and that you will respond favorably as part of the continuous improvement of your products and services.
> Do not hesitate to contact me if necessary.
> Kind regards

RenaudPerr, 2024-10-05:
HI all,
Believe me or not, I haven't got any answer from APS despite this message sent 3 times to 3 APS locations worldwide ;-))
Dear  @ksheumaker, if you agree with my improvement suggestions and if you have connections with APS staff / research & Dev teams, could you please forward them this message again please?
Kind regards
Renaud

HAEdwin, 2024-10-28:
I do not have connections with APS staff in fact APsystems seems not open at all for suggestions like providing release notes when new firmware is released. I tried in the past but also failed, I hope someday they improve the company policy on this issue.