ksheumaker/homeassistant-apsystems_ecur #268: Enhancement: Add inverter AC_power, per panel DC_voltage, DC_current

mi-hol, 2024-09-14:
It seems that currently APSystemsECU.py has some shortcomings.

From my view these are:

- AC_voltage is reported in per panel data, where DC_voltage would be expected. Further outlined in [this comment](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/30#issuecomment-1933060456)

-  AC_power for total inverter output is missing, but seems the most relevant value as AC_power is what gets charged by utility companies. Currently total DC input power is reported as "power" but that is a too high value due to DC to AC conversion losses.

-  per panel DC_voltage, DC_current are missing

From  findings of reverse engineering of APSystems ECU in https://community.home-assistant.io/t/apsystems-aps-ecu-r-local-inverters-data-pull/260835/238 I assume (but have yet to proof!) that above shortcomings could be fixed.
 
Unfortunately the fix could introduce compatibly issues for existing solutions. Eventually just a fix of "power" as "AC_power for total inverter output" & per panel DC_voltage could be feasible.

Therefore an alternative would be adding a new class with the new values (and eventually fix of "power" as "AC_power for total inverter output")

May I ask what course of action would be the best approach to address the shortcomings?

mi-hol, 2024-09-14:
Example from (restricted to installers) APS EMA report:

[image]

Data same time from (slightly modified) ECU_B.py:

```
{'current_power': 945,
 'ecu_firmware': 'ECU_B_1.2.33',
 'ecu_id': '2163******',
 'inverter_qty': 1,
 'inverters': {'7020********': {'DC_power': [469, 476],
                                'DC_voltage': [240, 240],
                                'channel_qty': 2,
                                'frequency': 50.0,
                                'model': 'YC60/DS3-S-M-D-L',
                                'online': True,
                                'signal': 73,
                                'temperature': 45,
                                'uid': '7020********'}},
 'lifetime_energy': 2001.8,
 'qty_of_inverters': 1,
 'qty_of_online_inverters': 1,
 'timestamp': '2024-09-14 13:14:41',
 'today_energy': 3.87}
Today energy : 3.87 kWh
Current power (AC): 945 W
Total energy : 2001.8 kWh
Number inverter: 1
Inverter Id: 7020********
Frequency: 50.0 Hz
Signal: 73 %
Temperature: 45 °C
Voltage inverter (DC) 1 panel 1: 240 V
Power inverter (DC)1 panel 1: 469 W
Power inverter (DC)1 panel 2: 476 W
```

Difference "Current power (AC): 945 W" to "actual AC power: 897 W" is 5.2% (= DC to AC conversion loss)

HAEdwin, 2024-09-14:
So this goes for an ECU-B with an YC600?

mi-hol, 2024-09-14:
for an ECU-B with a DS3 and after this works with a QT2.

Most important from my view is an answer to question:

> May I ask what course of action would be the best approach to address the shortcomings?

HAEdwin, 2024-09-14:
The query retrieves other data fields than what is being sent to EMA. The query in the integration does not contain DC voltages from individual panels (afaik). The data pushed to EMA does so that's why you see them in the EMA site.
Where you mention: "AC_voltage is reported in per panel data" you mean AC voltage per inverter right?

mi-hol, 2024-09-14:
> Where you mention: "AC_voltage is reported in per panel data" you mean AC voltage per inverter right?

correct

HAEdwin, 2024-09-14:
So that data is correct, I mean - I only see AC voltages that are within the standard for the country ~230V AC in The Netherlands

HAEdwin, 2024-09-14:
These are the data fields (YC600) which are being sent to EMA (currently zero because the inverter is down)
[image]
It's taken from the proxy solution (under contruction): https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy

mi-hol, 2024-09-14:
> It's taken from the proxy solution (under contruction): https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy

the data I'm after is apparently extracted in api.py function get_inverters.
Do you happen to know to APS command that triggers this response?

HAEdwin, 2024-09-15:
Unfortunately not. What we've done in the beginning is looking and interpreting data dat is shown in the ECU app (later EMA manager) and there it does not contain the data you're after. There is no documentation or further reference on what commands can be send to the ECU.

mi-hol, 2024-09-15:
> looking and interpreting data dat is shown in EMA manager and there it does not contain the data you're after. 

The data is available in EMA web but only shown to privileged "installer" accounts (see screenshots in https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/268#issuecomment-2350959248 showing the data I'm after)

HAEdwin, 2024-09-15:
Yes but it ain't in the apps (unless you discover something else) and therefor is not included in this integration. The proxy method intercepts data that is sent to EMA and it does include the data.

HAEdwin, 2025-01-29:
I'll close this issue for now, do take a look at the new version: https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader
If you want DC voltages you might want to try the proxy at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy

feel free to add a new issue there if you need.

closed 2025-01-29 by HAEdwin