ksheumaker/homeassistant-apsystems_ecur #269: Total amount of energy in HA is 7% more than value in EMA app

worker72, 2024-09-15:
I have an ECU-B with two DS3 inverters. I managed to get the integration installed and connect it to the ECU-B. But there is something strange with the Total energy that is reported. 

The value of ECU lifetime energy in the HA integration is 255kWh
The Total Energy value in the EMA app is 239kWh (and by the way: this is the same value I see when I go to the website of EMA)

This is not incidental, I have been looking at this for some time now. The difference between the two is slowly increasing.
It's not a huge problem, but I'm very curious to know where this difference comes from.

worker72, 2024-09-15:
I tried to "download diagnostics" but my browser then says "file wasn't available on site"

HAEdwin, 2024-09-15:
Yes is is a known issue but not due to the integration which is just reading ECU values. There are more people questioning where the difference is but until now no one knows.

HAEdwin, 2024-09-15:
> I tried to "download diagnostics" but my browser then says "file wasn't available on site"

You first have to enable the integration's debug mode... if you stop debugging it will ask automatically if you would like to download.

HAEdwin, 2024-10-21:
Part of the difference between EMA en HA seems to be caused by the fact that EMA uses the timestamp contained in the received data which runs 5 minutes behind and HA uses the current timestamp. Afaik HA is not able to "post data in the past" i.e. correct data that is posted in the past.

HAEdwin, 2025-01-29:
I'll close this issue for now. Might dig deeper into this later but it's a fact that EMA does it's own calculation based on data sent.
Feel free to open the issue again at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader

closed 2025-01-29 by HAEdwin