ksheumaker/homeassistant-apsystems_ecur #270: APSystemsInvalidData exception: Result on 'Inverter data' incorrect end signature '202' != END data=b'4150533131303031353030303230320a

pvtex, 2024-09-27:
Hi

i had to replace my ECU-B with a new one.
Used the replace ECU function in the app.

But now i need to add the new ECU to HA.
When i try i get this error message:

```
Traceback` (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 42, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 120, in query_ecu
    data = self.process_inverter_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 224, in process_inverter_data
    self.check_ecu_checksum(data, "Inverter data")
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 175, in check_ecu_checksum
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Result on 'Inverter data' incorrect end signature '202' != END data=b'4150533131303031353030303230320a' 
```

HAEdwin, 2024-09-27:
First thought is that it's just a message, received data does not contain the expected END signature. Might be caused by all kinds of things like firmware upgrade. Leave it running for a while and see what happens. You can also put the integration into debug mode, that might give some additional info.

HAEdwin, 2024-10-28:
Closing issue assuming it was a temporary data glitch and not due to the integration.